pbatard/rufus #1657: Application Freezes when open with usb stick plugged

andrelelis-ff, 2020-11-30:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
[35520] Rufus x86 v3.13.1730 
[35520] Windows version: Windows 7 64-bit 
[35520] Syslinux versions: 4.07/2013-07-25, 6.04/pre1 
[35520] Grub versions: 0.4.6a, 2.04 
[35520] System locale ID: 0x0409 (en-US) 
[35520] Will use default UI locale 0x0409 
[35520] SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E 
[35520] Localization set to 'en-US' 
[35520] Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387) 
[38520] [38520:40772:20201130,184133.374:ERROR directory_reader_win.cc:43] FindFirstFile: The system cannot find the path specified. (3)
[32424] [3032:core_112421621120146564544] sync_engine.cc:1661:operator() Sync engine activity state:
[32424] operation_queue_size: 0
[32424] all_downloads: 0
[32424] operation_queue_unique_size: 0
[32424] change_ids_up_to_date: true
[32424] largest_local_change_id: 3442
[32424] fetching_changes: false
[32424] 
[32424] [17624:ipc_thread] ipc_socket_win.cc:277:GetNextConnection Accepting next connection

```

andrelelis-ff, 2020-12-03:
> Are you using Sophos Antivirus by any chance (or any of the incompatible sofwtare [listed here](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software))?
> 
> Almost all of the reports I've had of this issue boil down to Sophos doing a lousy job at allowing legitimate applications such as Rufus to access USB flash drives, and this looks like a duplicate of #1597.

Yes, It is just like that. Will check, thanks for flag!

pbatard, 2020-12-10:
No further update, so I will assume it was Sophos. &rarr; Closing.

closed 2020-12-10 by pbatard