pbatard/rufus #1660: GB, MB, KB - small fix for Russian Language (maybe not only)

PaulNevada, 2020-12-08:
Hello, Author!
In Russian it is Gb, Mb, Kb and that's not right... Maybe, not with one language only... Thank you very much!

pbatard, 2020-12-08:
So you're saying the Russian abbreviation for GigaByte should be Gb and not GB?
If that is the case, what is the Russian abbreviation for Gigabit (as in network speed for instance)?

PaulNevada, 2020-12-08:
No! It is fow now is incorrect - Gb, Mb, Kb... :)))

PaulNevada, 2020-12-08:
When a flash drive is selected, in English it appers ok - 64 GB, but in Russian (haven't checked other languages) it shows up 64 Gb and that is wrong...

pbatard, 2020-12-08:
Hmm, this is what is being used right now: https://github.com/pbatard/rufus/blob/master/res/loc/po/ru-RU.po#L287-L323
So I'm not seeing `Gb` being used but `Гб`. Is that not correct?

Or is it for the website? But there too I am seeing things like `Мб` being used.

Here's what I see in Rufus:

[image]

And here's what I see for the website:

[image]

So where exactly are you seeing `Mb` or `Gb`? And are you sure it isn't the _potential fonts missing on your side_ (which is an issue on **your** computer that has nothing to do with either Rufus or the web site) that is the issue?

PaulNevada, 2020-12-08:
By Gb I meant Russian Гб...

pbatard, 2020-12-08:
So what should it be???

PaulNevada, 2020-12-08:
> So what should it be???

Should be ГБ, МБ, КБ. The second letter should be high as it is in English.

pbatard, 2020-12-08:
Okay. So `Б` is the capitalization of `б` in Russian. Okay, I'll try to fix that when I get a chance.

PaulNevada, 2020-12-08:
Yes. Exactly. Thanks!

maykimayk, 2020-12-08:
dil programı ekleme

commit 8bb31f6a (2020-12-11): [loc] fix Russian translation for KB/MB/GB...

* Closes #1660
* Also update GPT partition types and set UTF-8 codepage in .rc
closed 2020-12-11 by pbatard