pbatard/rufus #1664: Grub2 MBR

Thinstation, 2020-12-16:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am trying to make an ISO that can be written with Rufus to thumbdrive, that covers all the boot positions with grub. Right now, rufus is not detecting my grub install. I searched through the code and discovered that my files need to be in /boot/grub/i386-pc, and called grub.cfg and grldr. I discovered I also needed to copy normal.mod into that location as well. When I load the ISO in rufus, it only offers UEFI mode, not BIOS, which is a needed bootable mode. Am I doing something wrong here?

Log
---
```
Rufus x86 v3.11.1678 (Portable)
Windows version: Windows 10 64-bit (Build 18363.1198)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'SDXC Card' (0BDA:0306)
Using autorun.inf label for drive F: 'ThinStation'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7795, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.7 GB (64121470976 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ThinStation'
  Size: 65.5 MB (Projected)
  Uses: EFI
Using image: thinstation-efi.iso (65.9 MB)
```

pbatard, 2020-12-16:
Please update to Rufus 3.13. You're using Rufus 3.11 where an issue with GRUB is present, that was fixed in later version.

As a general rule, if you encounter an issue, you should see if a newer version is available and try with that one.

You can download Rufus 3.13 from https://rufus.ie.

As I am pretty sure your issue will be fixed once you upgrade, I will close it.

Thinstation, 2020-12-16:
It didn't, but thanks for the update. The message changed slightly, but my grub install is not detected, and I only have the UEFI option.

Rufus x86 v3.13.1730 (Portable)
Windows version: Windows 10 64-bit (Build 18363.1198)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SDXC Card' (0BDA:0306)
Using autorun.inf label for drive F: 'ThinStation'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7795, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.7 GB (64121470976 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ThinStation'
  Size: 65.5 MB (Projected)
  Uses: EFI
Using image: thinstation-efi.iso (65.9 MB)

Thinstation, 2020-12-16:
Forgive me, I'm not super swift with the C, but how exactly is rufus trying to detect the version of Grub? It looks like it's parsing normal.mod for a string instance. This file doesn't appear to be critical to the actual boot process.

pbatard, 2020-12-16:
Hmm, then whoever created that image modified GRUB in a manner that makes it incompatible with Rufus.

Rufus must know the version of GRUB being used because different versions are incompatible with one another, and, if you compile GRUB from its official source, the version **will** be present in `normal.mod`.

So, someone (was it you?) went out of their way to remove that version, and supporting images that purposefully remove standard bootloader stuff is always a lose-lose situation, both for the developer and the user of the application. Therefore I'm afraid that you're on your own there (though the image may work in UEFI mode).

If you are creating your own ISO, please don't remove the version, as this only creates problems.

pbatard, 2020-12-16:
Also `grldr` is for GRUB4DOS, not GRUB 2.0, so if you are trying to combine GRUB4DOS and GRUB 2.0, you're only going to create more problems. I would encourage you to research how Linux distros compile their own versions of GRUB 2.0, an follow something similar to create your ISO.

closed 2020-12-16 by pbatard

Thinstation, 2020-12-16:
Yes I am compiling from source. I'm not doing anything other than standard stripping. I am following the fedora recipe. I was actually wondering that the significance of grldr was, as you can really name the file anything you want. Is it just searching normal.mod for a string begining with Grub version - (some number)?

pbatard, 2020-12-16:
https://github.com/pbatard/rufus/blob/master/src/iso.c#L1058-L1082 and https://github.com/pbatard/rufus/blob/master/src/iso.c#L1058-L1082.

Unless I'm starting to see mainline distros fail the current GRUB version detection (last time I checked Fedora, they were using Isolinux/Syslinux for BIOS boot), I'm not going to alter it, so it's up to you to make sure it works. Or you can have a look at what Ubuntu does, since they transitioned from Isolinux to GRUB 2.0 for BIOS boot, and everything works fine there.

I'll take a look at the most up to date Fedora to see if I need to change anything, but if Fedora is detected okay, then it'll be up to you to try to comply with what mainline distros do.

Thinstation, 2020-12-16:
Thanks Pete,

When I say I am using the Fedora recipe, I mean for compiling source into binaries. Fedora is still using isolinux/syslinux for legacy boot, but that project is basically dead. I'm finishing up the secureboot procedure for my distro, and decided to transition to grub for all booting. Thanks for letting me know about Ubuntu. I'll reverse that. Thanks for making rufus, it's a fantastic utility.

pbatard, 2020-12-16:
Okay, then, yeah, Ubuntu is probably where you want to look at, because they successfully transitioned to GRUB everywhere for BIOS and UEFI. Oh and just FIY, if you're wondering why Rufus doesn't simply try to ignore the GRUB version and install the most up to date version it has, the problem is that, when we tried it for Kaspersky images (issue #467), it just rendered the drive unbootable (which isn't surprising as mixing GRUB modules and a GRUB bootloader of different revisions usually only leads to broken boot).

Thinstation, 2020-12-16:
Also, thanks for finding those lines of code for me, but I'm still struggling to understand what exactly it's trying to do to get the version number. Also, that file isn't used for booting, what file does rufus try and install in MBR span?

pbatard, 2020-12-16:
You need a version that contains `GRUB  version ` (with 2 spaces between `GRUB` and `version` and one after `version`.

And the files used for BIOS boot are `boot.img` (what gets written into the MBR) and `core.img` (the secondary BIOS boot loader).

These are provided by Rufus, since you can't just take the GRUB boot loader used for loading an optical disk and use that for USB... which is why we need to know the version, so that, if you're not using GRUB modules (which come from the ISO and which we do reuse) that are compatible with the version Rufus provides, we need to download a version that matches your modules, and therefore we need to know the GRUB version being used, hence what we need to pick something that looks like a version in the base GRUB 2.0 module for x86 BIOS boot: `normal.mod`.

Thinstation, 2020-12-16:
Ty, hopefully that's enough to get me through this.

Thinstation, 2020-12-16:
What module set do you use for your grub-mkimage command?

pbatard, 2020-12-16:
The ones documented in the `grub-mkimage` command you can see at https://github.com/pbatard/rufus/tree/master/res/grub2 (or in the `readme.txt` from that same directory, which is what GitHub displays).

Thinstation, 2020-12-16:
Ty. I really appreciate you leading me through all the hooks here. I was able to track down the patch that breaks the test.
https://src.fedoraproject.org/rpms/grub2/blob/master/f/0030-Don-t-print-GNU-GRUB-header.patch
It's used by Fedora and Redhat. I'll see what grub looks like without it.

pbatard, 2020-12-16:
Oh great, Fedora and RedHat are going to make it impossible to support them in Rufus the day they switch to using GRUB for BIOS. Yet another great initiative by Red Hat, doing things that nobody asked for...

Thinstation, 2020-12-16:
LOL, I feel your pain.

I was actually gonna ask if you(I feel you cringing) could make a change to simplify things. I can easily generate all the image formats that are needed. I just need rufus to pick up that a file in a special place and name is what should be writen to the MBR gap.

pbatard, 2020-12-16:
I'm **EXCEEDINGLY** reluctant to add features to Rufus that are only there to serve someone's custom workflow, because then it opens the floodgates to _"But you added code for that guy!!! Why can't you add code for me???"_, and I receive more than enough requests for custom workflows already...

Now, you can always tell precisely what you would like to have. But I make absolutely no promise you will get what you want.

Thinstation, 2020-12-16:
I get it. I totally get it. Not a lot of people have made the jump completely over to grub. It's funky, but I'm starting to like it. There is no real spec at this point, but I do have a few ideas that should work with what Ubuntu has done so far. 

they are placing all there loader pieces in /boot/grub/<platform>/ , but they are only including grub images for efi. I'm guessing this why you are downloading your own build of grub. What if I placed a file at /boot/grub/i386-pc/ called boot.img, and if it's found, skip normal.mod and just write boot.img?

Thinstation, 2020-12-16:
I really don't know why Ubuntu includes the pieces of grub, but not final grub binaries. I would actually put the binaries in and skip the pieces.

Thinstation, 2020-12-16:
Or maybe you want to do it more like syslinux, where you have a downloaded files section, and I can place the boot.img there.

Thinstation, 2020-12-16:
It looks like Ubuntu changed it up a little between 20.04 and 20.10. I like 20.10 with the way they moved the font and final binary down to /boot/grub.

Thinstation, 2020-12-17:
As to why I would want to build my own bootloader vs taking yours, I refer to lines 339-343 of iso.c. I intend to embed an early config file that tells each binary for the various boot methods which way it's coming in, so the OS knows whether it's persistent or not. Its also more consistent with EFI, where I can control everything just by where things exist.

Thinstation, 2020-12-17:
Thanks to your excellent documentation, I was able to hack my own grub boot.img and core.img into rufus. I guess I have to get a signing key now, but that's easy enough.

Thinstation, 2020-12-17:
Sorry to keep tagging you, but why does my custom build of rufus fail to complete writing the removable media? Is it because it's unsigned?

pbatard, 2020-12-17:
> Sorry to keep tagging you, but why does my custom build of rufus fail to complete writing the removable media?

No idea. Most likely because of whatever changes you applied to the source.

> Is it because it's unsigned?

I'm writing GPLv3 software, so the last thing I want is whatever people build from source to be different from my builds. I may insert a message that tells you you are running and unofficial or ALPHA/BETA build, so that I don't end up having to support stuff that I shouldn't be supporting, but that's about it. I will certainly **not** cripple the functionality of the application in any way, if you happen to build your own (unsigned) version. 

The whole point of the GPL, as opposed to other licenses, is that you can't just go around and keep a monopoly on your software, by placing strategic elements under closed source to prevent other people from producing their own indistinguishable clone of the original, which:
1. Forces you to compete in a level playing field and ensure that you do a better overall job than would be competitor(s)
2. Ensures that, the day you give up on maintaining the project or abandon it altogether, and someone is motivated enough to pursue it, ownership will organically shift to the newcomer since they will be able to pick up from exactly where you left, with no feature missing from the original work.

As such, it would be very crummy of a developer to add stuff like crippling the app if it detects that the signature doesn't match their own. If someone was inclined to do that, they'd never use a GPLv3 license.

So, to answer your question more succinctly: No.

Thinstation, 2020-12-17:
Ty. I didn't think it was something that Rufus was doing as much as Microsoft. My build writes about 50% before failing, and I thought it might have been a security violation on the windows platform.

pbatard, 2020-12-17:
Signatures are validated before the app launches. There's no validation that occurs when an application is running, even if the executable is elevated.

Now, if you are writing ESPs, you should know that Windows is a complete ass when it comes to that, and may remove that volume from under your feet (since Microsoft decided users should not see an ESP, ever!) while you are writing to it. I've actually been struggling with trying to keep Windows off my back on that up to the current version of Rufus.

If you are not dealing with ESPs, and you're not doing anything special, the most likely culprit would be a security solution, such as the ones that scream as soon as they see an `autorun.inf` file (even a benign one). Or it could be one of the many poorly designed apps that are adamant on keeping a file handle on a volume that they don't use (such as Paragon ExtFs for Windows). I have a **non exhaustive** list of these little pests [here](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software).

Thinstation, 2020-12-17:
It was my core.img. It's slightly larger than yours, but it looks like the code is calculating size, so I'm not sure what's happening....

pbatard, 2020-12-17:
Oh yeah, you may have an issue if your `core.img` is larger than 31.5 KB **if** you are using _Fixes for old BIOSes_, because then the first partition will be created at the 32 KB mark (assuming 512-byte sectors). But Rufus should warn you about this or prevent the use of "Fixes for old BIOSes" altogether. And if you don't use that options, the first partition is at the 1 MB mark, leaving plenty of space for whatever oversized `core.img` you may have (since `core.img` gets written into the "free" space after the MBR).

Other thing I can think of is that you copied `boot.img` and didn't alter the matching hex data in `src\ms-sys\inc\mbr_grub2.h` which is what Rufus actually uses, though I doubt you did anything that would alter the GRUB2 MBR (which has been pretty much unchanged for the past 10 years). But this is mentioned in `readme.txt`.

Thinstation, 2020-12-17:
I did replace boot.img as it is different, and I did update the hex in mbr_grub2.h. Took me hours to figure out how to do the conversion. One thing I noticed, is that if I truncate my boot.img to 432 bytes, the last 6 bytes are null while yours goes all the way up to 432, and I think 448 is allowed. 

I did /not/ enable "Fixes for old BIOSes" 

Grub has actually changed quite a bit over the years, but it's a weird project. The version string is meaningless. Fedora/Redhat is applying over 200 patches, and I doubt my .mod files are compatible with your build.

pbatard, 2020-12-17:
> The version string is meaningless

I beg to differ. Having a version is certainly something that has been needed for Manjaro in the past: https://github.com/pbatard/rufus-web/tree/gh-pages/files/grub-2.03.5

That's the whole point of being able to identify the version, so that, if a distro needs a specific GRUB version, Rufuc can fetch it from its server.

Which brings me to pointing out that, if you need a custom `core.img` to be installed by regular Rufus, all you have to do is have a `normal.mod` version that's unique enough (and matches the Rufus version lookup for GRUB), and provide me with enough details so that I can build `core.img` myself (as I will **not** upload any binaries that I haven't built myself or can be validated as coming from the official project, on the Rufus servers).

Thinstation, 2020-12-17:
When I say the version string is meaningless, I mean it just doesn't say enough about the condition of the code that is being compiled. Syslinux has official releases, and appends a custom build number to anything else, which is perfect. I can populate rufus_files with my custom ldlinux.sys and ldlinux.bss.

GRUB is the wild west. So far as I know there are not *official binaries in circulation, almost everybody patches the hell out of that project. If there were I could induce rufus to use my boot.img and core.img in the rufus_files like syslinux, that would be great.

pbatard, 2020-12-17:
Yeah, every distro is ending up having to apply 100's of patches on top of latest release, because the current GRUB maintainer(s) certainly seem to take their time releasing anything, even when there exist critical vulnerabilities that absolutely warrant a dot release.

I've [officially complained](https://lists.gnu.org/archive/html/grub-devel/2020-10/msg00073.html) about this, because it __is__ creating problem that go beyond distros having to maintain their own set of semi-incompatible patches, and creating an awful mess for everyone, but it doesn't look like the GRUB maintainers care or realise the issues their reluctance to release create.

If I were you, I would try to wait for GRUB 2.06 to be released, as it may reduce the amount of patches that need to be applied, though, considering that 2.06 was supposed to start its release process in 2020.02... and we've just seen a similar promise of an official 2.06 release coming soon [a couple days ago](https://lists.gnu.org/archive/html/grub-devel/2020-12/msg00215.html) (which, if you check the string of quoted dates, provides a sad overview of the status of GRUB maintainership when it comes to understanding that, no, you can't just delay a release for a year and pretend that everything is peachy)...

Doncuppjr, 2020-12-30:
So I am trying to wrote an core.img that is 230kb in size. I'm not using the Fixes for old BIOSes, but the only way I've been able to get it to work is by increasing that MAX_DATA_LEN in file.h. You suggested that I shouldn't have to do that.

>Oh yeah, you may have an issue if your core.img is larger than 31.5 KB if you are using Fixes for old BIOSes

Am I missing something?

pbatard, 2020-12-30:
> You suggested that I shouldn't have to do that.

Can you point me to the location where I appear to suggest that? Because I'm pretty sure I didn't do any such thing, when I pretty much said that I didn't see the point of raising the limit above 64 KB when the old BIOS compatibility requires a `core.img` that is smaller than 31.5 KB, therefore implying that if you do use a `core.img` that is larger than 64 KB, not only will you not be able to use the fixed for old BIOSes, but you will of course need to alter that limit.

What I said is pretty much: I set the limits to what I have need for, with regards to my official releases. I did not set them to accomodate other folks with regards to whatever customization they want to apply to their own releases. Therefore, if you use huge `core.img`, it's up to you to modify the limits.