pbatard/rufus #1666: No device detected (from Wine+Ubuntu)

VinceFINET, 2020-12-20:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
Running Rufus from Ubuntu and Wine (+ MFC*.DLL).
As seen in log, no device is detected.
When my USB stick is mounted correctly in the system
I tried once again with my usb stick unmounted, same.

Ubuntu version: Linux  5.4.0-58-generic #64-Ubuntu SMP x86_64 GNU/Linux
Wine version: wine-5.0.3

Log
---
```
Rufus x86 v3.12.1710
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: CoCreateInstance failed; hr = 80040154
Localization set to 'fr-FR'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
0 devices found
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_FR-FR_DV9'
  Size: 5.8 Go (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_20H2_v2_French_x64.iso (5.8 Go)
```

VinceFINET, 2020-12-20:
Same with latest version:
```
Rufus x86 v3.13.1730
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: CoCreateInstance failed; hr = 80040154
Localization set to 'fr-FR'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
0 devices found
```

JonnyTech, 2020-12-20:
Rufus is designed to run natively on Windows only: https://github.com/pbatard/rufus/wiki/FAQ#do-you-plan-to-port-rufus-to-linuxmac-ossome-other-os

VinceFINET, 2020-12-20:
I missed that part of the FAQ sorry 
Thank you anyway this is an awesome tool!

closed 2020-12-20 by VinceFINET