pbatard/rufus #1668: Suggestion: Do not hide the DOWNLOAD option when update checking has been disabled.

43v6c3463, 2020-12-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
The ISO download button/dropdown option should not be completely hidden with nothing but a note in the log about it missing when check for updates has been disabled. The option should stay visible but show a message about update checking having to be enabled for it to work. A lot of people use Rufus in the following manner: They download the portable version, choose No on the check for updates prompt, do their business and then delete it pending redownload for when they next need it. These people might not even be aware of Rufus having ISO downloading functionality due to the way that option is completely hidden without warning. I was one of these people back when the download feature was first added and I've met others too. I also have to tell people to pick YES on the update prompt when telling them about Rufus and how it can be used to create Windows 10 install media. Please consider changing this as this is literally the only complaint I have about Rufus, thanks.

Log
---
```
Rufus x86 v3.13.1730 (Portable)
Windows version: Windows 10 64-bit (Build 19042.630)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found non-USB removable device 'VMware Virtual NVMe Disk' => Eliminated
0 devices found

```

pbatard, 2020-12-28:
Well, the thing is, the tying of the availability of the download button to the check for update is quite intentional, because I consider that most people **will** enable the check for update, and that it's mostly paranoid people, who don't want an application to download and run anything remote, and especially not a PowerShell script (which is what Rufus actually does behind the scene when you press Download), that will choose not to enable the check for update.

As a developer, I very much want people to leave the check for update enabled, because it's common for me to fix a major bug from one version to the next, and I **really** don't want to have to rely on people downloading Rufus manually to get these fixes, especially as I usually release new versions on a semi regular basis and if folks reject the check for update, they might miss a critical update that happened soon after they downloaded.

Or, to put it differently, if you choose not to enable the check for update, then I consider that you already made a decision to have a **degraded** user experienced compared to the one the application was actually designed for (since I designed Rufus with the idea that people would have the check for update enabled, and I am only offering the option to disable it **as a standard courtesy** for people who are paranoid about a features [that I made sure could be trusted not to do something malicious](https://github.com/pbatard/rufus/wiki/Security#Update_signature)). So, besides the logic of considering that someone who doesn't want an application to contact the network for updates doesn't want the application to download and run a script from the network, it makes a lot of sense to me to tie further "degradation" of the UI to that choice.

Finally I do consider that the log message, as well as the [relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Help_I_dont_see_the_ISO_download_button), should be enough to help people who chose to disable the check for updates and are looking for the download option.

&rarr; Rejected.

closed 2020-12-28 by pbatard

pbatard, 2020-12-28:
Also, to further elaborate on your suggestion of having the download being there and displaying a message for people who have check for updates disabled, I don't think I can just go with a short _"You need to enable check for updates if you want to use this feature"_ popup (which is not a bad suggestion), because for the sake of transparency, I would need to inform people that the Download feature will download an run a remote PowerShell script, as I consider that folks who choose to disable the check for updates did so precisely because they don't want the application they run to do these kind of things. So this would require a rather lengthy message which (and I'm receiving regular feedback about this about the DD vs ISO mode popup) some people feel they are **entitled** to ignore if it's too long, and which also require translation work.

Possibly, if I were to do this properly, I might try to keep a short message and add a _"Please check the 'Update policy and settings' section for more details"_ in the dialog, and add an entry about the PowerShell script download in the Update policy and settings, but I see this as a lot of work for little benefit, so I don't see myself doing something like this anytime soon.