pbatard/rufus #1671: "Help, I don't see the ISO download button!"

reclaimed, 2021-01-02:
Hello, Peter! I appreciate your work A LOT. And still, there's a paragraph in FAQ which makes me a bit angry:

"Therefore, even if you have 'Check for updates' enabled, if there was an issue with your network while Rufus launching or if you are running Rufus in an unconnected environment, the ISO download button will not appear."

I mean WHY. Why wouldn't you let the user decide whether to select the option or not? It's not a public kiosk software to target it for zero-level users.

pbatard, 2021-01-02:
Well, there are a few reasons for that:

1. The Download feature downloads and runs a PowerShell script, from the internet, behind the scenes. And I consider that people who chose to disable the check for updates, despite its usefulness, on account that it will connect to the internet to fetch content without asking them (past the initial prompt), most likely don't want to have the application connect to the internet to fetch other content. You have to understand that, if I were to have a prompt to ask folks whether they want the Download button to be enabled, I'd have to explain what it does behind the scenes, which, as I've seen with the ISO vs DD dialog, some people feel entitled to ignore if it's more than a couple lines (_"TL;DR"_), and then deal with the possibility that someone might want the Download feature but not the Check for update feature, which means multiple dialogs on startup (or a single lengthy one with a set of options that users aren't too sure about), and users complaining that Rufus is interrupting them with choices they don't care that much about when all they want is use the application. So, **in my view**, it remains much better to collapse the choice of enabling the Download button into the check for updates, to avoid yet another choice that a lot of people will either not read properly or ignore, especially as I consider that most sensible people will enable the check for updates anyway.
2. As a software developer, I very much want people to have the check for updates enabled, because I worked long and hard on that feature to ensure [it was secured](https://github.com/pbatard/rufus/wiki/Security#Update_signature), and I want to be able to push critical bugfixes ASAP **to help users**! As such, I consider that people who disable the check for update are mostly paranoid ones, and therefore will **logically** not want to enable a feature that downloads and runs a remote script. And if I have to nudge Rufus users into enabling the check for updates so that they can get critical fixes as soon as they are published, rather than whenever the heck the user feels like it, by tying it to other features, then so be it. And I know it may sound to some as somewhat autocratic but please remember that you get the application entirely for free so if I, as its developer, have do do something that goes against some people's wishes, because I consider it will lead to a better situation for users overall, then that's generally what I'll pursue. In other words, I'd rather continue publishing software that doesn't ask for financial compensation, and still have the final word with regards to its design so that I can apply what I consider to be best for users (even if some users fail to see it that way and will insist that I'm doing everyone a major disservice instead) than feel coerced into applying changes that I don't feel actually benefit the User Experience.

 
And that is why the Download button is tied to the Check for updates, and why I am not planning to change that anytime soon, even if it appears to makes some users angry. If you don't like that, then please remember that the application is fully Open Source and that you're of course free to create your own version.

closed 2021-01-02 by pbatard

Hrxn, 2021-01-03:
You know why this is kinda funny to me?
Because both of you are entirely right.