pbatard/rufus #1673: иммеются проблемки при записи при помощи Rufus

anonims, 2021-01-07:
записываю win10 Версия: 20H2 на флешку - выкаченный при помощи программы Rufus с сайта microsoft  win10 Версия: 20H2 или же беру образ со своих проверенных источников
- после выбора устройства с которого начать установку win10 пк долго ожидает, а потом просто перекрашивает экран в чернозеленые цвета, но увы инициализация процесса установки так и не происходит
- использовал 3 разных флешки и 4 разных пк везде одна и таже проблема, может кто сталкивался с проблемой?

pbatard, 2021-01-07:
This is a Windows installation question, not a Rufus question. Please ask on Microsoft Windows user forums.

closed 2021-01-07 by pbatard

anonims, 2021-01-09:
не соглашусь с данными решением) ибо ваша программа rufus версии 3.13 уже не может верно записывать win10 Версия: 20H2 для чего она собственно и создавалась, а это уже говорит о багах в программе