pbatard/rufus #1676: Keep speed comparison between Rufus and other applications up to date

Ostrjs28, 2021-01-10:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
On the current Rufus github and Rufus official website there is speed comparison between Rufus and other applications.

Unfortunately the speed comparisons are outdated and have not been updated for almost 9 years.

It would be very good if it could be tested with current hardware and current Windows 10 ISO (20H2).

Log
---
```
<FULL LOG>
```

pbatard, 2021-01-10:
Duplicate of #1153. As you may understand this is very low priority for me and overall, I'd rather have an updated comparison carried out by someone who might be seen as obviously biased towards a specific application.

closed 2021-01-10 by pbatard

Ostrjs28, 2021-01-10:
@pbatard 
I don't think that Rufus users or users who want to test Rufus for the first time will find it bad if the Rufus developer himself updates the speed results again.
I think they like to see the actual speed results to get an approximate result of what you might expect. Personally, I would find it very cool.

You can do it, no one is angry or will attack you or anything.

pbatard, 2021-01-10:
Yes, if nobody else volunteers, I am still planning to do it. But I consider it low priority compared to enhancements and fixes that I see as a lot more beneficial for users.