pbatard/rufus #1679: USB drive write-protected after failing of rufus

JokeGamer, 2021-01-11:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

JokeGamer, 2021-01-11:
Hello and Happy New Year, yesterday I was trying to make my usb bootable (kali linux) as I wanted to double boot my Windows 10 with Kali Linux. After processing like 50% of it (or a little bit less) it said that device is already in use by another program and said it was Windows Explorer. Then I closed Windows Explorer and then started again, it gave me an I/O error, and when I'm trying to format it, it says Device is write protected. I tried using diskpart, registry, (no physical switch on USB) tried using EaseUS M, EaseUS partition master, another partition tool from bleeping computer, and AOMEI partition assistant. (I hope you (owner of rufus) on't get mad at me, I know I read FAQ in wiki that rufus can't damage USB.

pbatard, 2021-01-11:
> it gave me an I/O error

Which means that Windows detected that your hardware had an issue, and that this is the **cause** rather of the consequence of Rufus and other software reporting an error.

In other words, while you may be under the impression that because Rufus reported an I/O error, it created a hardware problem it's actually the opposite: A hardware problem happened on your media, **that was not caused by the software that was accessing it** and as a result, the software reported an error.

> I read FAQ in wiki that rufus can't damage USB

That is correct. However that does not mean that flash based USB drives won't fail.
Typically what happens, because the flash memory used in USB flash drives is a lot less reliable than people imagine, the flash memory will fail, even when the drive is left completely idle and is not connected to a computer. Then, the next time an application (and it can be any application) tries to write the flash memory cells that are defective, the flash memory controller on the USB detects that the flash has an issue and sets the drive to read-only to avoid further damage, and give you a chance to read the existing data. When that happens, Rufus (or Windows, or Etcher, or whatever) will report an I/O error as you try to write to the device, even though they had nothing to do with the failure.

As such, I'm afraid there is not help I can provide. [As per the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media) a flash drive that switched to read-only or that reports no media is a drive that is defective and that should be replaced with a new one. And there's not much help I can provide here, since these issues happen naturally and do not stem from using this or that specific application (which, as Rufus did, will simply report an I/O error when they happen).

As there is not much I can help you with, since you will need to replace your media, I will close this issue.

closed 2021-01-11 by pbatard