pbatard/rufus #1680: Rufus 3.13 or older -  Slow writes and "halts" when creating from Windows ISO (GPT / no CSM)

Karl-WE, 2021-01-12:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [?] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus has become very slow when creating Windows 19xx / 20xx ISOs to very fast and even new USB drives. In the past i was hardly able to read any individual files except boot.wim and install.wim / esd.
I have attached two videos demonstrating the issue.
I can 100% assure this is no local hw issue as tested on complete new hardware aswell as other older hardware using different USB drives with same results. I suspect it started with a later 3.x release as the first releases worked as expected.

Log
---
```
0 devices found
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {9FC59847-C239-489C-A3B9-C4DC1B049A5C}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {9D53C847-A714-4E21-9CF3-D779F207447D}
  Size: 14.9 GB (16012877312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 14.9 GB)
Opened \\.\PhysicalDrive4 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.9 GB)
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{9d53c847-a714-4e21-9cf3-d779f207447d}\
Successfully remounted \\?\Volume{9d53c847-a714-4e21-9cf3-d779f207447d}\ as D:
Extracting files...
Image is an UDF image
Extracting: D:\boot\de-de\bootsect.exe.mui (19.5 KB)
Extracting: D:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: D:\boot\fonts\cht_boot.ttf (3.7 MB)
[... 900 lines cut ...]
D:autorun.inf already exists - keeping it

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {F67318BE-3060-42B6-AC52-CEC29D649E31}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {008C1AE9-6FCB-4AAF-82C8-BAEF0EE3EE40}
  Size: 14.9 GB (16012877312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
  
 Computing checksum for 'F:\Windows 10 20H2 x64-de-de.iso'...
  MD5:    7dfca5662291f858ab283d0c781ddf95
  SHA1:   978d2afbc3185a2af74f0e78464ccd39e1b85275
  SHA256: 90a976897bc347abd04cccb04b297cc6c4cddbf2304fc20b7c4e4eba07469e9b

```

Link to videos and native log
https://1drv.ms/u/s!ApTx3d3fhinPhLEM9ljnxYLFnlC9GA?e=gp9Z78

pbatard, 2021-01-12:
> Rufus has become very slow when creating Windows 19xx / 20xx ISOs to very fast and even new USB drives.

_"has become"_ should be easy to demonstrate. Please run the same test with any of the older versions you can find at https://rufus.ie/downloads/ and see if you actually see a speed improvement with these older versions.

Also make sure to run these test with the real-time scan of your antivirus disabled, as a specific antivirus may decide that a third party program enacting a copy warrants more scrutiny than a copy being enacted by the OS UI.

> In the past i was hardly able to read any individual files except boot.wim and install.wim / esd.

That is *not* a good measure of loss of speed. You are describing perception, and I'm afraid I will need some actual verifiable proof, from your testing with older versions of Rufus. If what you describe is true, it should be very easy for you to find an older version that seems to behave much better in terms of speed than the latest version, and back up your claim with something a bit better than your perception that individual files seem to display for longer.

Furthermore, Rufus does measure the actual time it takes to create a media, and of course older versions of Rufus are still compatible with newer Windows ISOs, so you shouldn't even have to produce videos to back up your claim, but should just be able to report the total creation time reported by Rufus.

This is what I am currently observing with the older Rufus 2.18 using `Win10_20H2_EnglishInternational_x64.iso`:
[image]

And this is what I am observing with Rufus 3.13 (latest) using the same image:
[image]

Of course, these tests were conducted against the same USB drive (An almost brand new SanDisk Extreme Pro USB 3.1 128 GB).
So, even if this was punctual testing (I have only ran a couple of tests), I cannot validate your claim that Rufus has gotten slower, as I am observing pretty similar times between 3.13 (`1:20`) and 2.18 (`1.29`).

This is consistent with my expectations, since, despite what people seem to think, almost nothing has changed in terms of ISO extraction between 2.x and 3.x, and I have certainly not spend any development time trying to improve or alter that code.

> I have attached two videos demonstrating the issue.

Except you are comparing file copy from an ISO that was natively mounted by Windows against ISO extraction from Rufus, which proves nothing with regards to Rufus having gotten slower.

The ISO extraction of Rufus has more overhead than the speed you can achieve through direct file copy using Explorer as, because mounting an ISO is not a feature that is available on Windows 7 or earlier, and we needed to extract ISO content on that platform as well as Windows XP, we had to implement our own ISO extraction support, for which we are using libcdio. 

Now, unlike some other ISO extraction utilities, I believe that libcdio is starting to show its limits when it comes to being able to feed data as fast as current fast USB drives can process, because it isn't really optimized for high throughput (unlike 7-zip for instance), and, whereas up to recently we could count on USB drives being slow enough for that not to be a factor, when using much faster drives, we're starting not to fill Windows buffer fast enough to reach maximum throughput. Plus, at the moment, we're not parallelizing the reading and writing of the data, as we could do, which would probably help improve the write speed.

So I do have plans to revisit the way Rufus extract content, because I will be the first to say that the landscape has changed enough in the past few years to warrant some improvements.

But I will dispute that Rufus has "become" slower, and I am pretty sure that, if you do test with older versions, you will find that they are about as slow (or as fast) as the current ones...

Karl-WE, 2021-01-13:
Hi sorry despite the preparation, I should have displayed it more specifically. Opening an older version is really an eye-opener how much efforts you have put into rufus.

I think I have found the main culprit. 
What is still noteable that the progress bar is moving so consistent as you see in the video (stopping at certain points then going on)

20H2 - MBAM 3.3.0 Premium, Rufus 3.1.3
[image]

20H2 - MBAM 3.3.0 Premium, Rufus 2.1.1.995
[image]

20H2 - Windows Defender, Rufus 2.1.1.995
[image]

20H2 - Windows Defender, Rufus 3.1.3
[image]

Karl-WE, 2021-01-13:
so the main issue was MBAM. Sut what can we do that progress bar works more consistently? do you use Write-Back or Write-Through to copy the files? Just asking because I learnt about a major difference in Write-Back starting 1809.

Karl-WE, 2021-01-13:
here is a screenshot with 2.11 when I enable the previous state before 1903, so using the cache, but this also enables lazy write back introed in 1809

20H2 - Windows Defender, Rufus 2.1.1.995 
1:37 (20H2, fast removal policy since 1903)
1:55 (20H2, manual removal policy pre 1903, with caching enabled)
[image]

pbatard, 2021-01-13:
Yeah, Antivirus software and Windows caching can be both a blessing and a curse, as they sometime introduce very unexpected behaviour in what should be fairly straightforward operations (such as copying data from one file over to another file).

Again, that is not to say that there aren't steps that could (and should) be done in Rufus to improve copy speed, because for one thing, 7-zip manages to far outclass Rufus on ISO file extraction, especially for large files, even though, on principle, they should be accomplishing something pretty similar behind the scenes. So there is definitely room for improvement, starting with using asynchronous/OVERLAPPED I/O, which Rufus currently doesn't currently do on account that, when I tried it some years ago, against USB drives (that were of course slower than the ones we have now), [it didn't seem to make any improvement at all](https://github.com/pbatard/rufus/blob/3056d54cc1c1b18a91aa50088819cb0d350b3aca/src/format.c#L1562-L1564) over simply letting Windows caching take care of ensuring that the write buffers were never starved from standard synchronous I/O.

So I will definitely be spending some time in the future trying to see what can be done to improve Rufus ISO extraction speed even as what can be achieved there is still going to be conditioned by Windows' caching policies and overintrusive security solutions...

Karl-WE, 2021-01-15:
Thanks for your time @pbatard.

closed 2021-01-15 by Karl-WE