pbatard/rufus #1682: Additional copying is needed

DJAnthon, 2021-01-16:
﻿

- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Can you add ability to copy user defined folder to disk after image loading?
Sometimes image loading lasts 10-15 hours and it will be nice to do additional data copying without pause.
Thank you for your program.

Log
---
```
<FULL LOG>
```

pbatard, 2021-01-16:
The feature you request is already present.

1. Compress all the additional data you want to have added on top of your ISO into a `.zip` file. It must be of Zip type (i.e. PKZIP), not 7-zip (i.e. LZMA) or some other compression
2. Select your ISO as normal with the <kbd>SELECT</kbd> button.
3. After your ISO is selected, press <kbd>Ctrl</kbd>-<kbd>SELECT</kbd> to select your `.zip`

If you do this properly, you will find that the extra files from the `.zip` archive will be extracted along with the ISO content.

Obviously, this can only work when writing an image in ISO mode.

Also, if it takes 10-15 hours to write data, you have a **MAJOR** issue with the hardware you selected, so please look into using a computer that supports USB 3.x as well as a USB 3.x drive with a **write** speed (not read speed, which is useless) of at least 100 MB/s, which aren't that hard to find these days. And also make sure you don't have an application that is interfering with the write operation, such as an obtrusive security solution.

As I have answered your question, I will close this issue.

closed 2021-01-16 by pbatard

DJAnthon, 2021-01-16:
Great thanks, this will be enough for me. English is not my native, so I didn't read all instructions and FAQ, sorry.
About hardware, making flash loader is reason to fix problems with usb, hdd, ssd and other stuff, but sometimes damaged hardware can work in safe mode at very low speed and I fix them by remote control, so I have no choice sometimes. But Rufus is very convenient and helps me. Earlier I used another software that now don't work properly.