pbatard/rufus #1683: Rufus wont read my drive atfirst it was a 1tb NTFS, but I formatted it to fat32 and still no change. At this point im ready to give up, but you never know someone might be able to help me here.

Uhnitials, 2021-01-19:
﻿Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19042.746)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Processing Hub 1:
Processing Hub 2:
Processing Hub 3:
  Hub[1] = '\\?\usb#root_hub#4&18803e48&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[114]: USB\VID_1532&PID_0214\5&2950A0CE&0&4
  Found ID[196]: USB\VID_1E7D&PID_2E22\5&2950A0CE&0&5
  Found ID[256]: USB\VID_0C76&PID_161E\5&2950A0CE&0&1
Processing Hub 4:
Processing Hub 5:
Processing Hub 6:
Processing 'IDE' device:
  Disabled by policy
Processing 'IDE' device:
  Disabled by policy
0 devices found
Processing Hub 1:
Processing Hub 2:
Processing Hub 3:
  Hub[1] = '\\?\usb#root_hub#4&18803e48&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[114]: USB\VID_1532&PID_0214\5&2950A0CE&0&4
  Found ID[196]: USB\VID_1E7D&PID_2E22\5&2950A0CE&0&5
  Found ID[256]: USB\VID_0C76&PID_161E\5&2950A0CE&0&1
Processing Hub 4:
Processing Hub 5:
Processing Hub 6:
Processing 'IDE' device:
  Disabled by policy
Processing 'IDE' device:
  Disabled by policy
0 devices found

if000else, 2021-01-21:
Oh, shit, the same issue....

Uhnitials, 2021-01-21:
> Oh, shit, the same issue....

The reason why is because rufus only shows removable media, like your USB. Rufus can only install isos on usb drives and some smaller usb hdd/SSD's. Hope this helps you out

closed 2021-01-21 by Uhnitials