pbatard/rufus #1684: promoting a feature request on the Windows Setup Deployment Team

Karl-WE, 2021-01-21:
Dear Pete, 
how about if the ever growing WIM files would be splitted by Microsoft in their ISOs?
They have SWM but not using it for unclear reasons. ESD costs more CPU time on installing / compressing
We are going to hit the 4GB marker even on ESD files, but NTFS is not compatible with Secure Boot.

This guy nails it. 
 https://techcommunity.microsoft.com/t5/windows-it-pro-blog/simplifying-on-premises-deployment-of-servicing-stack-updates/bc-p/2079506/emcs_t/S2h8ZW1haWx8dG9waWNfc3Vic2NyaXB0aW9ufEtLNVEwNzdWRUMyOUNTfDIwNzk1MDZ8U1VCU0NSSVBUSU9OU3xoSw#M2315

I am aware that you did an awesome implementation to circumvent the current issues and later rufus version make clear disclaimers for UEFI. 

If you think it is good, please support my feature request here, as I see it exactly the same way as Joerg.
here is the feedback hub link

https://aka.ms/AAawx6w

pbatard, 2021-01-22:
> but NTFS is not compatible with Secure Boot.

NTFS __IS__ compatible with Secure Boot.

The only issue is that Microsoft refuses to sign GPLv3 code, for completely abusive reasons (if you need a more lenghty explanation [than the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Why_do_I_need_to_disable_Secure_Boot_to_use_UEFINTFS) as to why Microsoft's "reasons" are complete bullshit, you can read [this](https://github.com/pbatard/uefi-ntfs/issues/20#issuecomment-739441450)).

If the exact same for the NTFS driver was licensed under GPLv2 instead of GPLv3, UEFI:NTFS (and the NTFS driver) would long have been signed for Secure Boot, and we wouldn't be having this discussion.

As such, I'd much rather have users complain to Microsoft that their "no GPLv3" policy for Secure Boot signing is creating grief than consider that it's perfectly acceptable for a large corporation to abuse the heck of their position to arbitrarily decide what license people can and cannot use for Secure Boot signing. I therefore don't see why **I** should have to make an effort to work around what is a plain and simple abuse of power.

As long as people like yourself don't report that the GPLv3 signing issue is preventing them from using what is by far __best__ solution for > 4 GB files (because every other solution requires reworking the ISO content in some way), the situation will not change, because if you agree that Microsoft are entitled to entrench themselves in their position, as abusive at it might be, then you must agree that I am similarly entitled not to invest time working on implementing another solution, when I already have a perfectly valid one.

So please complain to Microsoft. Thank you.

closed 2021-01-22 by pbatard

Karl-WE, 2021-01-22:
Hi Pete,
please chill :) i was not aware about such politics and as a developer of free software I can certainly understand your frustration of Microsofts action or no action.
However I did not open this as a complaint, quite opposite hoped that solve an issue, from your point of view solved by other ways, and hoped for your support for the idea.

Karl-WE, 2021-01-22:
what you think of this blog I have received from a SCCM pro.
https://miketerrill.net/2016/11/21/create-multiple-partitions-on-any-usb-flash-drive/

pbatard, 2021-01-22:
I understand that this wasn't a complaint.

But I am getting frustrated from getting request after request asking me to implement whatever "alternate" solution people want to point me to, that should ensure compatibility with Secure Boot, when the solution I already developed works just fine, and is compatible with Secure Boot (as a matter of fact I don't even have to modify the code of UEFI:NTFS to ensure that only Secure Boot signed drivers are loaded and Secure Boot signed bootloaders are chain loaded)... except for an arbitrary decision from Microsoft, that does not stand scrutiny.

I would therefore very much prefer if, instead of asking **ME** to do something about a problem that I have **already** solved, people asked Microsoft to drop their abusive no GPLv3 Secure Boot signing policy. I should not have to perform extra work on something that, as far as I am concerned, has already been more than properly solved. It is an utter waste of my time.

> what you think of this blog I have received from a SCCM pro.

I think that you need to realize that, even if I agreed to go the trouble of splitting content over 2 partitions, I need Rufus to remain compatible with Windows 8.x (and preferably Windows 7, as some people may precisely be wanting to use Rufus on obsolete Windows 7 to upgrade to Windows 10), which means that the _"Starting in the Windows 10 Insider Preview build 14965..." is a complete deal breaker.

I've already discussed this matter at length, with previous folks who, like yourselves, suggested the multi-partition approach, and, again, at this stage, considering that I have to stress out that it is not me who should bear the burden of the remaining Secure Boot compatibility issue, but Microsoft, I will kindly ask that you go bother Microsoft if you feel like following up on this.

Karl-WE, 2021-01-22:
Thank you Pete for taking time to express your point. Much appreciated wasn't aware this is a recurring request. If so perhap you might want to pin this closed PR as a "warning" for others nagging you about this remotely. I allowed myself to post the good intention of post yesterday and added it to my FR in the FBH so that the windows setup team is aware about this GPL issue, perhaps they are, perhaps they are not. But we have flagged it now for sure in their prefered tool.