pbatard/rufus #1685: Error:[0xC0030090] The directory is not a subdirectory of the root directory

cpu2020, 2021-01-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When i try to flash a parrot linux os onto my 1tb usb with 100gb persistence i get the error Error:[0xC0030090] The directory is not a subdirectory of the root directory"

Log
---
```
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/20191223/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 866.8 GB)
● Partition 2 (offset: 930716059648, size: 100.0 GB)
Skipping GUID volume for '\Device\Floppy0'
Opened \\.\PhysicalDrive6 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 866.8 GB)
● Creating Linux Persistence Partition (offset: 930716059648, size: 100.0 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
1638399 possible inodes out of 26214394 blocks (block size = 4096)
1310719 blocks (5.0%) reserved for the super user
Creating 800 inode sets: [1 marker = 10.0 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Failed to create ext3 'lost+found' dir: Corrupted entry
Re-mounted volume as F: after error

Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB SM Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
Found USB 2.0 device 'Generic USB MS Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB CF Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
1 device found
Skipping GUID volume for '\Device\Floppy0'
No volume information for drive 0x86
Disk type: Removable, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 126207, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x005166AF
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 866.8 GB (930715011072 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 100.0 GB (107374159872 bytes)
  Start Sector: 1817804804, Boot: No
Closed Windows format prompt
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB SM Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
Found USB 2.0 device 'Generic USB MS Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB CF Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
1 device found
Skipping GUID volume for '\Device\Floppy0'
No volume information for drive 0x86
Disk type: Removable, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 126207, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x005166AF
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 866.8 GB (930715011072 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 100.0 GB (107374159872 bytes)
  Start Sector: 1817804804, Boot: No

```

pbatard, 2021-01-22:
Please provide the __FULL__ log, as requested, or I will close this issue. I am missing tons of important data because you chose to truncate the log. The checklist is very explicit about this:

> - [ ] The log I am copying is the FULL log, starting with the line Rufus version: x.y.z - I have NOT removed any part of it.

Yet, even though you checked it, you did NOT follow the instructions. Please do so if you want to receive support, thank you.

cpu2020, 2021-01-22:
sorry about that i dont think i copied it correct here:

Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19042.746)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: OpenLocalMachineGPO failed - error 80004005
Localization set to 'en-US'
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB SM Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB MS Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB CF Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
Device eliminated because it was detected as a Hard Drive (score 5 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB SM Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB MS Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB CF Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
Device eliminated because it was detected as a Hard Drive (score 5 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB SM Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB MS Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB CF Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
Skipping GUID volume for '\Device\Floppy0'
No volume information for drive 0x82
Disk type: Removable, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 126207, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00A337D0
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 866.8 GB (930715011072 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 100.0 GB (107374159872 bytes)
  Start Sector: 1817804804, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Parrot security'
[... 62 lines cut ...]
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB SM Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB MS Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB CF Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
Skipping GUID volume for '\Device\Floppy0'
No volume information for drive 0x82
Disk type: Removable, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 126207, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0001408E
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 866.8 GB (930715011072 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 100.0 GB (107374159872 bytes)
  Start Sector: 1817804804, Boot: No

pbatard, 2021-01-22:
Thanks. I will look at this when I get a chance.

cpu2020, 2021-01-22:
Ok Thank you

cpu2020, 2021-01-22:
i have also tried running this on a vm today and still gives the same error i tried a diffrent version aswell and still gives the same error

cpu2020, 2021-01-25:
hi again,

ive worked out this only happens when i try to add persistence and only happens on large usbs (bigger than 64gb) 
not sure if it will help at all but thought id let you know

thanks,
cpu

pbatard, 2021-02-03:
I'm unable to replicate this issue with a 100 GB persistent partition on a 120 GB USB drive:

```
Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19042.746)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found UAS (USB 3.0) device 'Mushkin Ventura Ultra SCSI Disk Device' (174C:55AA)
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {9F48F37B-FF31-42F9-A789-BAE2962B9AA4}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  ID: {ED323CD9-8954-4B84-B5C6-3C9CDD348E1B}
  Size: 447 MB (468713472 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ParrotSec'
  Size: 3.6 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: Parrot-security-4.1_amd64.iso (3.5 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 447 MB)
Opened \\.\PhysicalDrive5 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 11.8 GB)
● Creating Linux Persistence Partition (offset: 12659948032, size: 100.0 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
1638399 possible inodes out of 26214394 blocks (block size = 4096)
1310719 blocks (5.0%) reserved for the super user
Creating 800 inode sets: [1 marker = 10.0 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 65536 journal blocks: [1 marker = 819.2 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Created 'persistence.conf' file
Done
Formatting (Large FAT32)...
[... 974 lines cut ...]
Created: F:\syslinux.cfg → /isolinux/isolinux.cfg
Updating F:\md5sum.txt:
● /boot/grub/grub.cfg
Finalizing, please wait...
F:autorun.inf already exists - keeping it

Found UAS (USB 3.0) device 'Mushkin Ventura Ultra SCSI Disk Device' (174C:55AA)
Using autorun.inf label for drive F: 'Run Debian GNU/Linux'
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x19C8CD4C
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 11.8 GB (12658899456 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 100.0 GB (107374159872 bytes)
  Start Sector: 24726461, Boot: No
```

I'll see if I can test with a 1 TB drive, but it might take a while...

cpu2020, 2021-02-03:
strange,
the issue is still there for me

pbatard, 2021-04-26:
Tested again on a 2 TB disk. Still can't replicate the problem, even after setting various sizes for the persistent partition.

I'm going to have to close this issue, as there's only so much I can do if I am not able to replicate it, and it doesn't appear to be affecting many people.

closed 2021-04-26 by pbatard

thoth-stck, 2021-05-03:
Damn same issue brother did you manged to fix it?!!!

thoth-stck, 2021-05-03:
[image]

thoth-stck, 2021-05-03:
change it to NTFS