pbatard/rufus #1686: UEFI: Press any key to boot from USB.

dentix1, 2021-01-25:
**Checklist**
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [X ] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

**Issue description**

Is it possible to add "Press any key to boot from USB" when running in UEFI?
-------------------------------------------------------------------------------------
I have tried to create the USB with diffrent methods.
Tried with dual uefi/bios mode enabled, and with only UEFI/GPT.

I have searched FAQ and issues, and didn't any anything related to this issue.

pbatard, 2021-01-25:
1. You're going to have to explain what makes you think you need that, and, more importantly, how you think that this will benefit a large enough number of users, especially considering that you're the very first person to request such a feature.
2. Please point me to a (non custom) ISO that has a _"Press any key to boot from CD/DVD"_ UEFI bootloader, because I am not aware of any. My take is that adding the feature you request is really the job of the people who created the ISO (or, in the case of Linux, the GRUB/Syslinux developers) and not the one of Rufus, and that there's no reason to have that feature of USB if it doesn't also exist for CD/DVD boot. As is the case of the _"Press any key to boot from USB"_ that I added in Rufus **for BIOS/Legacy mode** because it existed for **for BIOS/Legacy mode** Windows bootloaders, I will need to see a regular ISO do the same for UEFI.
3. Unlike what is the case for BIOS/Legacy, if I were do add something like this for UEFI, then I'd have to get that bootloader Secure Boot signed (since, unlike UEFI:NTFS, there would be no licensing issue as this shouldn't be complex enough to warrant the reuse of GPLv3 code), which is a rather cumbersome and costly process for something that, so far, looks like a feature **you** would like to have, but that very few people should actually require. So, once again, I'd have to see some very heavy justification as to why you think I would need to do that in Rufus.
4. What general user-facing problem are you trying to solve here (apart from the "User may not be competent enough to know how to alter boot order in their UEFI settings", which I will dimiss as something that is not for Rufus to solve, as I've always designed Rufus with the expectation that its users will have a minimal level of computer proficiency)?

If this is just a "It would be nice if Rufus could..." feature, then I'm afraid I'm simply going to have to dismiss it, as per [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

dentix1, 2021-01-25:
1. Reason i'm wondering if it's possible is to prevent boot loop's on automated windows installation, if usb is selected as boot priority #1.
2. Microsoft deployment toolkit created iso's has "Press any key to boot from USB" it uses "efisys.bin" from windows ADK.
3. I'm not sure if efisys.bin could be used
4. Alot of users are used to windows 7 installation were it had "press any key to boot from usb" If its not possible to add they are just gonna get used to use "boot meny" instead.

I was just wondering if it was possible, I could try to add it, if you could give any hints on what to do. Any help would be helpful, thanks.

pbatard, 2021-01-25:
1. I think that Windows updates the UEFI boot order on first reboot to set the boot target to the destination disk (though I haven't confirmed this). One of the advantages of UEFI over BIOS is that the OS should (usually) be able to update the boot order on the fly. In other words, I'm not sure I get why Microsoft would go the "Press any key to boot from USB" route when booting in UEFI mode, as there should be a much better way to ensure that Windows will boot the installation disk after first reboot (and Windows definitely sets itself to the top of the boot order once installed, so you might as well do that early).
2. According to [this](https://docs.microsoft.com/en-us/troubleshoot/windows-server/deployment/create-iso-image-for-uefi-platforms), `efisys.bin` appears to be _a binary floppy disk layout of the EFI boot code_, so that should be a FAT image containing a UEFI bootloader.
3. Can't, because it's Microsoft copyrighted code, which I can't include in Rufus. That's the reason why, start with Windows 10, Rufus can no longer create MS-DOS bootable disks, because, since we can't redistribute Microsoft copyrighted content, we had to rely on the presence of the MS-DOS file in the system (they used to be included in `diskcopy.dll`), in order to copy them, but Microsoft removed that DLL in Windows 10. So you can't pick anything from `efisys.bin` and include it in Rufus, unless there's an agreement with Microsoft that anybody can redistribute that content, which I doubt there exists.
4. Again, using UEFI boot order should make the need for "press any key to boot from medium X" obsolete.

> I could try to add it, if you could give any hints on what to do.

Well you're going to craft a chain bootloader similar to https://github.com/pbatard/uefi-ntfs/blob/master/boot.c. Except the only way I'm going to accept that in Rufus, since it should be simple enough not to depend on pre-existing GPLv3 code, is if it's Secure Boot signed, because, unlike UEFI:NTFS, there's no legal or technical hurdle to get that signed. But the thing is, since https://github.com/pbatard/uefi-ntfs/blob/master/boot.c __is__ GPLv3, you can't actually reuse anything from that source, as any derivative work will have to be GPLv3... which Microsoft won't sign. So you're going to have to reinvent the wheel though, if needed, I can look into relicesing the GPLv3 UEFI:CODE from `boot.c` to GPLv2+ (which Microsoft will sign) as I wrote most of it, and I'm hopeful that, if I contact them, the other couple of people I picked code from would also agree to relicense their section of code (NB: unfortunately, it's not the `boot.c` license that's the issue to getting UEFI:NTFS Secure Boot signed, but that we also need to Secure Boot sign NTFS and exFAT UEFI drivers, which come from GRUB and can be relicensed. Otherwise, I'd have relicensed the UEFI:NTFS bootloader to GPLv2+ a long time ago). And, like what I'm doing for UEFI:NTFS, I'm going to need **all** of the arm/arm64/ia32/x64 versions of that bootloader, as I want people running Rufus on any of these platforms (for which I actually do provide a native Rufus version in the `.appx`) to create media that will boot on any other. But if you have something that works for x64 it shouldn't be that difficult to get that for the other platforms.

The most annoying (and expensive) part, if you go that route then, will be to get your boot loader signed for Secure Boot, though, on a technical level, I think it should be simple enough for Microsoft to validate that the code is safe for signing, as there shouldn't be much to it. Oh and for reference, `LoadImage()` will not load anything that doesn't pass Secure Boot validation in a Secure Boot enabled system, so you don't even have to specifically code your chain loader for Secure Boot, as that part is taken care of automatically as long as you use the UEFI system calls.

Now, of course, the other option is to rely on a media that already has a  "Press any key to boot from USB" chain loader, and use that, though I'm curious with regards to how that chain loading is supposed to happen (since, on x86_64, that chain loader would replace `/efi/boot/bootx64.efi` and therefore the initial ``/efi/boot/bootx64.efi` must be renamed to something else that the chain loader will chain boot. But, since it comes from Microsoft, I'd expect that chain loader to be Secure Boot signed.

For the record, if the media actually contains an `efisys.bin`, I could do something similar to what I'm doing to process `efi.img` that some Linux distros use (the principle is the same, as both these files are a FAT image of UEFI boot loaders). See https://github.com/pbatard/rufus/blob/master/src/iso.c#L1351-L1435 and https://github.com/pbatard/rufus/blob/master/src/iso.c#L1437-L1566, so most of the code to extract existing EFI bootloaders should be there.

But it still doesn't look to me like there's a common problem here, that users are struggling with, that warrants the effort at this stage. Especially, even as I do monitor social media sites like reddit and SuperUser for Windows installation issues, I don't recall ever seeing an instance of _"Darn, my Windows UEFI installation keeps booting back into the USB on first reboot"_, whereas, if "Press any key to boot from USB" was solving an actual problem, I'd expect to see tons of these.

[... cut ...]  I try to ensure that the target disk is clean (in which case you would of course get the desired behaviour of first boot to USB and second to HDD/SSD even if you don't fiddle with the boot order), I don't really remember an instance of Windows rebooting into the USB media post the initial file copy phase. So I would still very much first like to see some reports that this feature might indeed be useful when using the retail Windows installation ISOs one can download off the Microsoft servers...

pbatard, 2021-02-03:
As I believe I made my position clear on this, in that I don't see it as something that many users require, I will close this issue.

closed 2021-02-03 by pbatard

haseakash, 2021-03-05:
So can u please share how to add "Press any key boot from usb"  for  UEFI BOOT?

i need file code or etc.
please reply

haseakash2008@gmail.com