pbatard/rufus #1687: F2FS support for persistent partitions

Tachi107, 2021-01-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I removed things that do not apply since this is not a bug

Issue description
-----------------
Currently Rufus when creating persistent partitions it uses the `ext3` filesystem. I think that `F2FS` would be another valid (if not better) filesystem to use, especially because Rufus is manly used to flash images on devices that use flash memory (USB sticks). 
Would this be easy enough to implement?

pbatard, 2021-01-26:
> I think that `F2FS` would be another valid (if not better) filesystem to use

I think that people who are worried about `F2FS` vs `ext#` should be proficient enough to know how to switch/create their persistent partitions using `F2FS`, if they are that concerned.

Moreover, the Linux folks know that there are tons of people who are going to use `ext` on flash based media, and therefore modern OSes tend to take provisions for that when needed (for instance, distros are moving logging, which is something that can create a lot of wear and tear regardless of the file system you use, to a much more efficient disk backed RAM file system).

My take on this is that, if you are going to run Linux off a flash drive with a persistent partition, then you are unlikely to be running a database from it, but instead use persistence for non heavy writes. Because if you do need heavy I/O, the problem is not what Rufus will pick as the file system, but that you somehow went with persistence.

In other words, it's not the job of Rufus to **police** the ill-advised decisions people make if they **choose** not to perform a minimum level of research with regards to what they are trying to accomplish. Because if we go that route, people are going to expect me to turn Rufus in a virtual AI, that sits behind the user at every step of the way to advise them, on all kind of non related matters...

This should only ever matter if you use low grade media or a media that's ill suited for the job (if you are going to do heavy I/O from a persistent persistent partition, just shove an NVMe SSD into a USB enclosure), which also boils down to someone having deliberately made a poor decision on account of not having done their research.

> Would this be easy enough to implement?

No.

It was painful enough to add `ext#` support (it literally took **months**), and I had to extensively rework the e2fsprogs code to make it work with Windows. And again, I'd rather people who are paranoid about the suitability of a file system with regards to flash drives (I've got the same issue with some folks advising that I should switch to using exFAT for Windows, because they expect that NTFS will somehow destroy drives on account of being slightly heavier on I/O).

The only way I'm going to look into this is if someone comes with hard evidence that `ext#` is really bad for flash drives. But if such evidence exists, then I'd say you're probably better off forwarding it to the `ext#` file system folks, because it means that they did a poor job accounting to the wide range of media a file system might be used for, and that designers of a file system **must** provision for (and yes, even if new media types appear after the initial design of the file system).

&rarr; Rejected.

closed 2021-01-26 by pbatard

Tachi107, 2021-01-26:
Thanks for the detailed explanation, much better than anything I expected.
I agree with you, and based on what you said it is certainly not worth it.

Keep it up with the great work!