pbatard/rufus #1688: Feature request: Add button mnemonics

dancojocaru2000, 2021-01-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
If possible, I suggest adding mnemonics to the buttons of Rufus (letter in text is underlined, button can be "clicked" by Alt+letter). This will particularly help people using Narrator, as the narrator reads the keyboard shortcuts of a button. Pressing, for example, Alt+S for Start instead of navigating through each UI element can save quite a bit of time when you have to listen to the description of every UI element in order to find the right one.

Log
---
```
Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19041.746)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found UAS (USB 3.1) device 'Samsung Portable SSD T5 SCSI Disk Device' (04E8:61F5)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.1) device 'Samsung Portable SSD T5 SCSI Disk Device' (04E8:61F5)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found

```

pbatard, 2021-02-03:
That's a good suggestion. I'll see if I can implement that, but I'm afraid this will be low priority, and it may have to wait until the next UI redesign (which may be **years** away).

FlailAway, 2021-05-07:
That should be a simple issue within the Compiler. With the Button Caption Text, add an Ampersand at the access-character.

e.g.
"Start" make that "&Start" for Alt+S

"File Load" make that "File &Load" for Alt+L

"Create" make that "Cre&ate" for "Alt+A

If you need an Ampersand as real text, then use "This && That" for "This & That"

JonnyTech, 2021-05-07:
@FlailAway remember that Rufus is available in many languages which will all need vetting in case of key clashes, etc, hence the low priority.

pbatard, 2021-05-07:
Yes, there was a Pull Request going in that direction, but it's not that simple and I had to reject it, for the reasons [explained here](https://github.com/pbatard/rufus/pull/1689#issuecomment-773240275)...