pbatard/rufus #1692: Why i find 10000000000 viruses on virustotal.com?

M0nsterRED, 2021-02-12:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Why I am uploading Rufus-3.13p.exe on virustotal.com and see 100000 viruses in it's

Log
---
```
<FULL LOG>
```

pbatard, 2021-02-12:
Unless you are ending up on [this page](https://www.virustotal.com/gui/file/ec3136b053bd1559ad7ec1ea104113898093b886bf519e6117b138ef2e691cbb/detection), and therefore, unless the SHA-256 for your `rufus-3.13p.exe` executable is exactly `ec3136b053bd1559ad7ec1ea104113898093b886bf519e6117b138ef2e691cbb`, the version you have downloaded in __not__ an official version of Rufus 3.13.

Now, if you are relating to this actual page, then I hope you can agree that 100000 viruses  is a complete hyperbole.

The only **ALLEGED** virus that is detected by MaxSecure **AT THE MOMENT** is `Trojan.Malware.107817109.susgen` which is a false positive that I expect to be cleared shortly.

For the record, due to the nature of Antivirus scanning and the type of operations that Rufus performs, if you scan an application like Rufus by literally more than 50 different engines, you are bound to get one at some stage that will **mistakenly** detect something. It happens all the time, so much so that I have a [dedicated FAQ entry for that, which I would advise you to follow if you want to push the matter further](https://github.com/pbatard/rufus/wiki/FAQ#Antivirus_X_reports_that_Rufus_contains_malware).

Also, if you are implying that the additional snort reports from the same page are an indication of Rufus doing something malicious, there again, I will ask you to get a human security researcher to corroborate these claims before trying to blame the application.

Or, if you want to take what the VirusToral website reports at face value, you may also want to explain why the community score for this very same executable is positive, whereas, and as opposed to the scanning "engines" some of the people who apply this score are actual security researchers, who have a vested interest of ensuring that an application that is actually malicious gets a negative score.

Finally, I also invite you to read our [comprehensive Security Page](https://github.com/pbatard/rufus/wiki/Security).

So, before I close this issue, let me get one thing clear: If you want to get angry at someone for a "100000 viruses" report, then please get in touch with the people who are producing these dodgy/false positive reports, because they are the **only** ones to blame here and I'm getting quite fed up of having to justify the **FACT** that Rufus is safe to people who seem to be under the impression that if there's a couple of false positives on VirusTotal, then it must mean that I am a malware author, and treat me as such.

closed 2021-02-12 by pbatard

M0nsterRED, 2021-04-02:
> Unless you are ending up on [this page](https://www.virustotal.com/gui/file/ec3136b053bd1559ad7ec1ea104113898093b886bf519e6117b138ef2e691cbb/detection), and therefore, unless the SHA-256 for your [...]

https://www.virustotal.com/gui/file/ec3136b053bd1559ad7ec1ea104113898093b886bf519e6117b138ef2e691cbb/relations

pbatard, 2021-04-02:
The `/relations` page means just that, "relations". In other words, if I create a `.rar` archive that contains both `SUPER_MALICIOUS_VIRUS_TROYAN_FILE.exe` and `notepad.exe` (where `notepad.exe` is the official signed version from Microsoft), you will find that listed in relations for the innocuous `notepad.exe` executable on the VirusTotal page.

Does this mean that `notepad.exe` is a Virus? __NO__.

Well, the same applies for Rufus. Someone created a malicious archive that happened to __also__ contain the perfectly safe Rufus executable. Therefore, since VirusScan scanned that archive, and found two executables in there, it will list that malicious archive in the relations of whatever executables it found.

But the thing is: You don't have to "blindly believe" what I'm telling you here. You can just do a modicum of research, to understand what the `/relations` tab of VirusScan is all about, and come to the same conclusion.

The __only__ page that matters in the VirusScan report is https://www.virustotal.com/gui/file/ec3136b053bd1559ad7ec1ea104113898093b886bf519e6117b138ef2e691cbb/detection (i.e. `/detection`, __not__ `/relations`), which, at the time of this post, indicates that no AntiVirus has detected anything malicious in the Rufus executable.

But please, if you want to look like an idiot, please keep posting a link to the `/relations` page as if it was showing that Rufus is malicious, and thus demonstrate to the world that you clearly have no clue how to process a VirusTotal report...