pbatard/rufus #1694: download of windows image fails with error 'The operation has timed out'

stubhead, 2021-02-16:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable): [--NOT APPLICABLE--]
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
after selecting a windows 10 image from the list of those retrieved by the application and pressing 'confirm', rufus eventually returns w/the above error. pasting the offending url from the log into a web browser, i found that microsoft are sending a page asking that the install language be specified. i tried appending several common possibilities as a subsequent argument to the end of the url [lang, langId, language, languageId], but the error persists.

Log
---
```
Rufus x86 v3.13.1730
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Canon MG5200 series USB Device' (04A9:1749)
Device eliminated because it appears to contain no media
0 devices found
Downloading https://github.com/pbatard/Fido/releases/download/v1.18/Fido.ps1.lzma
File length: 7.4 KB
Successfully downloaded 'Fido.ps1.lzma'
Signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/en-US/software-download/
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=getskuinformationbyproductedition&sessionId=221af7b9-29e0-421c-a3c9-a687708e2285&productEditionId=1384&sdVersion=2
Error: The operation has timed out.
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=getskuinformationbyproductedition&sessionId=221af7b9-29e0-421c-a3c9-a687708e2285&productEditionId=1384&sdVersion=2
Error: The operation has timed out.
```
[image]

pbatard, 2021-02-16:
Network connectivity issues are rarely a problem from the application but more an issue with the user environment.

I'm afraid I am not seeing any issue with the downloads on my side:
```
Querying https://www.microsoft.com/en-US/software-download/
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=getskuinformationbyproductedition&sessionId=d84a5041-7446-4f77-aaf5-57a1b0e47ac1&productEditionId=1882&sdVersion=2
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=cfa9e580-a81e-4a4b-a846-7b21bf4e2e5b&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=GetProductDownloadLinksBySku&sessionId=d84a5041-7446-4f77-aaf5-57a1b0e47ac1&skuId=11608&language=English International&sdVersion=2
Exited download script with code: 0
Downloading https://software-download.microsoft.com/db/Win10_20H2_v2_EnglishInternational_x64.iso?t=84e60209-b26f-4b9f-b561-c4caa2835dc6&e=1613572089&h=91ae5c3bf9818f6f10cf583270d86029
File length: 5.8 GB
```

As such, I have to dismiss this report as a pure environmental issue. Please make sure you try from a different computer and preferably a different network before you report connectivity problems.

closed 2021-02-16 by pbatard

stubhead, 2021-02-17:
no, this wasn't a connectivity issue. as i mentioned in the issue report, the server at the address required a language id, which apparently hadn't been transmitted in the request from the app. if you simply type the url into a browser, you'll see what i mean. the timeout is due to the app waiting for a response from the server while the server is also waiting for input from the app -- not a network problem. in fact there were no network problems on my end at all, neither from my pc nor from my laptop.

i managed to find another source for windows iso's, so this isn't a pressing issue for me. however, you may want to take a look at it in order to bulletproof the app for future users.

pbatard, 2021-02-17:
> if you simply type the url into a browser, you'll see what i mean.

These requests are time-limited and become invalid after a while (that's why they have a session ID). And Microsoft's servers may very well IP restrict them.

So, yeah, the URL is expected not to work after a while, and that has nothing to do with the app.

Again, I am seeing no issue whatsoever using the same configuration as the one you describe (Windows 10, 19H1, Home/Pro), which I have just re-validated:
```
Querying https://www.microsoft.com/en-US/software-download/
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=getskuinformationbyproductedition&sessionId=2dcb315e-2c80-4be8-a90f-9330bd9602d6&productEditionId=1384&sdVersion=2
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=cfa9e580-a81e-4a4b-a846-7b21bf4e2e5b&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=GetProductDownloadLinksBySku&sessionId=2dcb315e-2c80-4be8-a90f-9330bd9602d6&skuId=8830&language=English International&sdVersion=2
Exited download script with code: 0
Downloading https://software-download.microsoft.com/db/Win10_1903_V2_EnglishInternational_x64.iso?t=0df2373f-664e-43ad-ace4-206d81cd6237&e=1613607716&h=ef1e700e1ca5fbd63f25fa12f841a53e
File length: 5 GB
```

Unless you can provide more details as to the specifics of what you are doing, that may lead someone carrying the exact same operation as the one you describe to actually experience the same error as you, I see nothing for me to "bulletproof", and, especially in the absence of any other similar report (if this was a common issue, as you assert, and with Rufus being downloaded tens of thousands of time a day, one would certainly expect to see more than a single report) I have to continue to dismiss your problem as purely environmental.

pbatard, 2021-02-17:
I would also strongly encourage you to try on a Windows 10 platform, as I am no longer supporting the usage of Rufus on Windows 7.