pbatard/rufus #1695: Download button discoverability suggestion

JVimes, 2021-02-17:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
I think the Download button discoverability could be improved with one of those question marks with a tool tip, like this:

[image]

Log
---
```
Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19042.804)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found UAS (USB 3.0) device 'HGST TOURO S SCSI Disk Device' (4971:1023)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1882, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x6601930C
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.4 GB (15482224640 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2021-02-17:
I believe this will create more confusion that it will solve.

I'm sorry but if you **chose** to reject the check for updates, which is opt in, then I consider that you obviously don't want to execute a remote script, which is what the DOWNLOAD button does behind the scenes to be able to do its job.

Therefore, I am not interested in trying to cater for people who seem to be paranoid about Rufus performing a benign update check (even as they clearly do trust the app to run elevated), but then seem to have absolutely no concern about running __a remote PowerShell script__ (that I am securing like crazy, yes, but that still remains a remote script) the minute it's about downloading an ISO...

As far as I am concerned, once a user has rejected the check for update, they have indicated that they made their **choice** about not allowing Rufus to download and execute additional data, and therefore it makes little sense to start providing hints about the DOWNLOAD button, especially as doing so properly and ethically would mean being absolutely explicit that the DOWNLOAD feature will download and execute a remote script before it displays the dialog (which is not really something that's suitable for a tooltip).

So I'm afraid I have no plans to make the tying of the DOWNLOAD button to the check for update more explicit by dropping hints in the UI, even more so as any change to the UI requires more than 35 translation updates, and I am deliberately trying to limit those.

&rarr; Rejected.

closed 2021-02-17 by pbatard

JVimes, 2021-02-17:
@pbatard you misread my post entirely. Seriously.