pbatard/rufus #1696: Reduce Pete's stress

JVimes, 2021-02-17:
Pete, this will reduce the number of people you have to send your boilerplate to:

[image]

Hrxn, 2021-02-17:
What's not to understand there?

pbatard, 2021-02-17:
I'd rather these people enabled the check for update **on their own** because they trust that the application (or its writer) is not going to do anything malicious behind their back and also because they understood that it might be critical for an application to download an update, in case a major bug has been fixed (which has happened in the past).

_Enticing_ people to enable the check for updates simply because they want to access the ISO downloads feature, without realizing that this also entails downloading and running a remote script (in elevated more, what's more) is not the way to go.

Once again, if **you** chose not to enable the check for update from Rufus (even as you are running an app that you probably didn't check the source of in elevated mode) you've basically told **me**, its developer: _"Sorry, but I don't trust you"_

Obviously, that is your right, and one that I fully support since it's only ethical for an app to ask for user permissions when doing something such as connecting to the internet (else I wouldn't be adding theses prompts in the first place).

But from the moment you select _No_ on that prompt, you have to be consistent with **your** choice and there's just no logical argument you can make that, if the user had known that the choice of not enabling the application to connect to remote servers to merely check if an update is available also tied to the ability of the application to connect to remote server to download and execute a script from it, they would have chosen differently.

If anything, it tells me that users who chose no and then would change their mind just for the convenience of accessing ISOs clearly need to spend some time reflecting on the reason why they make their choices in the first place. And that's not something a tooltip will help with, as it'd only lead them to picking options according to __convenience__ without weighing in the consequences of their actions.

In other words, I don't have a problem deriving "stress" from helping people being properly ethical in their actions. That's actually part of the reason I decided to release my software under a Free license. Thus, if you make a choice, you should make an educated one and understand that you will also need to follow through its ramifications. And since this is an important lesson to learn, I'm afraid you're just not going to find a reader's digest version of it in the app, even if it would make both your and my life more _convenient_...

pbatard, 2021-02-17:
I'll also remind everyone that Rufus does provide some indication of why the download check is disabled in the log. But, in line with what's being stated above, this is not something I want to make it too easy to find...

closed 2021-02-17 by pbatard