pbatard/rufus #1700: No Drivers on correctly formatted USB stick?

kxlvin, 2021-02-25:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.

Issue description
-----------------
Hello, I am trying to install a copy of Windows 10 on a newly built PC. I am able to format a USB stick with a bootable version of Windows 10 and my PC detects and access the USB with no problem. However, installation will not proceed when it comes time to install a drivers. I can an error message reading: "No signed device drivers were found. Make sure installation media contains the correct drivers." I've tried with the latest version of Rufus and then Rufus 3.9 after seeing an issue with Windows 10 and the latest Rufus version.

I don't have a log error because nothing crashed or anything but any help would be appreciated. Thanks.

pbatard, 2021-02-25:
Rufus does not provide the drivers or actually **any** of the content that Windows uses during its installation process.

Instead, everything comes from the ISO you provided.

Therefore, if Windows complains that it couldn't find such or such driver or that the driver is not properly signed, then I'm afraid that this is a problem that is 100% external to Rufus and that you will have to check with the support forums of whoever provided you with the ISO.

It would actually be illegal for Rufus to redistribute Windows drivers, because Windows drivers, like any other Windows content, are copyrighted by Microsoft, and they can not be legally included in an application like Rufus. What this really means is that, as opposed to what you appear to believe, none of the drivers you have an issue with were actually provided by Rufus.

As such, I would encourage you to post your question to a Windows support forum, as this issue tracker is only for issues that have to do with the Rufus application.

closed 2021-02-25 by pbatard

JonnyTech, 2021-02-25:
@kxlvin: also, please make sure that you use an official Microsoft Windows ISO. Rufus has the ability to download the file, in case you were unaware. You did not state where you got your version from.