pbatard/rufus #1702: USB is not detected after an error

mihail-vladov, 2021-02-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [N/A ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [ x] Running Rufus on a different computer.
- [ N/A] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Hi, I run an old version 2.15.1117 to make a bootable USB. At the end of the process, an error was shown. Sadly I didn't make a screenshot. After that, the USB can't be discovered by Windows. I installed the latest version of the tool and it can't discover it either even after I set the checkbox on the "List USB drives". 
Could you help me with this?
-----------------
<Please describe your issue here>

Log
---
```
Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 18362.30)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
0 devices found
```

pbatard, 2021-02-28:
From your report, it looks like your USB is dead.

The flash memory probably failed as Rufus was writing to it, which is what caused Rufus to report an error and your device to become undetected. It's very common for USB Flash Drives to fail that way, as flash memory is a lot more unreliable than people realize and once it starts failing, the flash drive controller usually takes drastic actions such as declaring the memory accessible altogether or read-only.

Short of trying to plug your USB on a Linux or Mac computer, to see if it's being detected there, there's really not much advice that can be provided apart from replacing your USB flash drive.

mihail-vladov, 2021-03-02:
Hi @pbatard, thank you for the prompt reply. I guess I will have to get another one :)

closed 2021-03-02 by mihail-vladov