pbatard/rufus #1704: About a "Persistent partition" format

yito24, 2021-03-23:
Select the Linux ISO and set the Persistent partition size.
Then the Persistent partition will be ext3.
Is there a way to change this?
Can you improve it so that user can choose this format? 

----Log excerpt -----------------------------------
`Using image: CentOS-8.3.2011-x86_64-dvd1.iso (8.6 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 2.8 GB)
● Partition 2 (offset: 3055550464, size: 26.0 GB)
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 8.7 GB)
● Creating Linux Persistence Partition (offset: 9344192512, size: 20.1 GB)
Waiting for logical drive to reappear...
Closed Windows format prompt
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (**ext3**)...
330026 possible inodes out of 5280423 blocks (block size = 4096)
264021 blocks (5.0%) reserved for the super user
Creating 162 inode sets: [1 marker = 2.0 set(s)]
++++++++++++`

pbatard, 2021-03-23:
If you're talking about adding `ext4` support, I have no plans do do such thing. There are too few benefits compared to the work required, and the main usage of a persistent system should be absolutely fine with `ext3`. And if you really want `ext4`, you should be able to reformat the partition to `ext4` once your system is up anyway, as the hard part (making space for the partition and enabling persistence) has already been carried out for you,

You can force `ext2` to be used for persistence instead of `ext3` through a registry setting, but I'm going to assume that this is not what you want.

I'm going to need to see a really strong argument for which __specific__ features of `ext4` make it a lot better than `ext3` (because if it's only "slightly" better, it doesn't make a lot of sense to add it in the context of a persistent live partition) to change my position on not adding `ext4` support to Rufus.

closed 2021-03-23 by pbatard

yito24, 2021-03-23:
I use rufus on Windows.
I can also use NTFS on Linux with ntfs-3g
I want to format to NTFS.

pbatard, 2021-03-23:
I'm not aware of many distributions that allow NTFS as a file system for the persistent partition.

You do realize that if I were to allow NTFS as a persistent file system, because I have __no means to validate that the distro being selected by the user is actually compatible with NTFS__, they may end up in a situation where persistence does not work at all.

For that reason, just like it makes little sense to add support for `ext4` selection, it makes little sense to add support for NTFS. If you do want NTFS for your file system, and please understand that you are in a very small minority there, then this is something you can sort out after your distro has booted.

&rarr; Rejected