pbatard/rufus #1705: [Question / Feedback] Multi-partition support in light of new Windows 10 capabilities.

VasVadum, 2021-03-23:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Feedback / Question
-----------------
First off, I'd like to say that I shortened the Checklist to only the things relevant to the situation, to help you more quickly see that I did see the FAQ before I considered posting my feedback, and that I did try to look for similar posts recently that were open or closed.

So you have in your FAQ that you won't add multi-partition support because computers can't see the extra partitions anyway, yes? Well, Windows 10 recently at some point became able to do so. I was able to plug in a flash drive with multiple partitions and Windows instantly recognized all the partitions, and this drive had been created back when it was unable to see both. Back when I made it in Ubuntu. Therefor this issue that your FAQ brings up shows that it is now possible for Windows to see partitioned flash drives normally.

In my case, I was trying to make a flash drive with a 6GB partition for the Windows 10 installation system, and another partition for backing up a copy of Windows after I install it, a compressed copy of the disk, on a USB 3.0 32GB flash drive where I was also planning to put drivers and whatnot if I had the remaining required space. I even pre-formatted the drive thinking Rufus would simply install Windows onto one partition, however it didn't, and instead reformatted the entire flash drive. :/

Now that it is extremely easy to create and use multiple partition flash drives, I don't think it would be easier having to "do it yourself" anymore. I wouldn't even know how to. I tried the default Windows 10 Media Creation tool and it broke my partition and made it strange and messed up. I can't think of any other way to do it either.

So here I am, asking if you might reconsider this possibility at some point, due to the fact that things have changed regarding the contents of your FAQ related to this particular thing. But if the answer is still no, perhaps you can update the FAQ as to why, and remove the reasoning behind it being because Windows can't see partitioned flash drives. If I must, I'll partition my old USB 2.0 64GB flash drive into as many small partitions as possible as a proof of concept and show the screenshots.

pbatard, 2021-03-24:
> you won't add multi-partition support because computers can't see the extra partitions anyway, yes? 

Not exactly. It's more a mix of how many people would really benefit from this feature (From the requests I get, I assert very few, but of course, since you are one of the persons placing such a request, you might see it very differently), how complex it would be to add and maintain, whether it would create more issues with users now being confused about whether they should choose to add extra partitions or not (there are already way too many people complaining that they see Rufus as confusing/too complex, so the last thing I want is add more options that may go towards confusing people, even under advanced settings). And finally, it has to do with whether __all__ versions of Windows that are supported by Rufus (rather than just a few) would be able to support such a feature.

> Well, Windows 10 recently at some point became able to do so.

Yes I know. The 1703 edition, a.k.a. "Creators Update" added it.
This is __explicitly__ mentioned in the FAQ entry you read, with, I quote:

> that is unless you are using Windows 10 Creators Update or later

The part that you are glossing over is that, as much as possible, Rufus needs to be compatible with older versions of Windows, and having an application that has to tell older system users about missing functionality is both a pain in the ass to write and to maintain.

And also, [this other FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_does_Rufus_need_to_delete_all_partitionsvolumes_when_creating_a_drive) is very relevant.

> So here I am, asking if you might reconsider this possibility at some point,

I'm afraid that some point will not happen sooner than EOL of any version of Windows before 1703. And I will need to see a lot more weight on how such a feature would benefit a majority of Rufus users, rather than just a select few, before I consider it.

Please bear in mind that my development time is limited, and I have tons of features, that would benefit a lot more users, to add before I would even consider something like this. As such, I am always very reluctant to add a feature that ultimately can be summed up as _"Some user with a __very specific workflow__ (that most Rufus users would not have) thinks it would be nice if Rufus could solve the specifics of their custom workflow"_ And now you are falling into the very relevant points mentioned in [this other FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

Also, everything is "extremely easy" when you don't have to do it yourself. Always be weary to tell someone you want to do something that it should be "extremely easy". You'd be very surprised how many of these "extremely easy" features I've been asked to implement over the years are nothing but.

&rarr; Rejected. If you need a custom workflow with multiple partitions, then you will have to put your own work and sort this out yourself, sorry.

closed 2021-03-24 by pbatard

VasVadum, 2021-03-24:
I do realize that the development time is limited, I just thought I'd put forth the idea again since some of the information had changed and I thought I'd share my unique method of using the feature. As for easy, I didn't mean easy for you to implement a feature, I meant that, you can simply load up Disk Partition tool in Windows and make multiple partitions with right clicking, being easy to make new partitions now from the user's end.

I do have one other thought, to add even though you rejected it though that might be something potentially helpful in future consideration or some such, I dunno. I realize that to make these drives boot on certain machines like UEFI that you may need to have a tiny invisible partition that tells the system UEFI information or whatever, I think I've seen this before but I can't be sure. In formatting the entire drive you write your code to all of it leaving no space to resize an already active partition. But what if instead of making new options for resizing and setting partitions, you add a checkbox, that formats the drive to unused space waiting to be partitioned, and your software just creates the partitions it needs at the sizes it requires? Then the rest of the drive, is empty space ready to be partitioned by the user using whatever tools they have available to them, and all it adds to the interface is one checkbox "Only use needed partition space" (with a nice little tooltip that simply explains that it will allow user to create new partitions on the drive after completion) and you can then put your partitions at the very beginning of the drive after its been entirely blanked out, which I believe is required for the OS to load it as it looks for the first blocks, right?

I realize that will probably not be easy either, but, you mentioned that you wanted to make sure not to complicate the UI too much, and I had that idea for simplifying it down to a single checkmark letting the more advanced users take advantage of the extra space on the drive if they should want. You can slide that checkmark under the advanced options as well so its collapsed by default.

Again, I realize none of this is likely to happen. I just like to put forth my thoughts and ideas in case something ends up helping. I'm not even sure if a closed ticket notifies you at all of replies but, I just wanted to get it out there while I had the thought. I've worked with game devs on projects so I know how complex it can get to implement something which is why I try to be detailed and find out the other angles and workarounds when noodling around with an idea of mine to cover every avenue that may not have been thought of.

Regardless, your time is your time, and if I could help donate I would because you have a great tool here, unfortunately I'm unable due to disability.