pbatard/rufus #1706: Denied from making a bootable drive

Krrishdhaneja, 2021-03-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When i installed rufus to flash ubuntu on my usb drive firstly I can't see my usb in the dropdown ->
[image]

And when I checked the log I was getting an `access denied` warning and `A device was eliminated because it didn't report itself as a disk` So what can be the problem for this?

Log
---
```
<FULL LOG>
Rufus x86 v3.12.1710
Windows version: Windows 10 64-bit (Build 21292.1010)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'Kingston DataTraveler G2 USB Device' (0930:6545)
Could not open '\\?\usbstor#disk&ven_kingston&prod_datatraveler_g2&rev_pmap#001d0f0ccf5cba30b3160064&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 20.04.2.0 LTS amd64'
  Size: 2.7 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: ubuntu-20.04.2.0-desktop-amd64.iso (2.7 GB)
Found USB 2.0 device 'Kingston DataTraveler G2 USB Device' (0930:6545)
Could not open '\\?\usbstor#disk&ven_kingston&prod_datatraveler_g2&rev_pmap#001d0f0ccf5cba30b3160064&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
```

pbatard, 2021-03-29:
`Access denied` usually means that __external software__ (i.e. one that Rufus does not have the possibility to enact control on, or bypass, especially as, most of that time, these would be AntiVirus software solutions) is preventing access to the drive.

So, you will need to identify precisely which software is preventing access, and then make sure you configure it in a manner that lets Rufus access the drive.

You may start by looking at [this list from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software), knowing that it is far from an exhaustive list, and that more software exists that can prevent access.

Also, since you checked the box that states that you tested with a different USB drive, which I can only assume means that you also got the `[0x00000005] Access is denied.` message with it, this pretty much corroborates that your problem comes from external software, in which case I will close this issue (since this also confirms that this is not a Rufus problem).

closed 2021-03-29 by pbatard

Krrishdhaneja, 2021-03-29:
Hmmm I checked that list but i don't have any of that software enabled and anti virus was also off, so what could be the potential problem?