pbatard/rufus #1707: Add support for 7z image archives

MichaIng, 2021-03-29:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Request
-----------------
Currently Rufus does not support the selection of a 7zip achieve file as image source, while zip, bzip2, gzip and xz are supported. On Windows, 7-Zip is used quite widely and a Linux alpha version has released recently, so it would be awesome if it could be supported by Rufus as well.

Bled code mentions 7z [here](https://github.com/pbatard/rufus/blob/master/src/bled/bled.h#L32), but there is no unpack function/source for it. Also for contents of WIM images, 7zip seems to be used already, so probably the same mechanism can be used for source image archives.

pbatard, 2021-03-29:
It's a bit more complex than it looks, because the 7z usage used for WIM relies on 7-zip for Windows to have been installed (separately) by the user, whereas the 7z support we want from Bled should be internal.

Quite frankly, adding support for 7z in Bled looks like a high cost/low benefit prospect, so, at this stage, I'm not sure I'll ever get around adding it, knowing that most of the compressed images I see would not be `.7z` ones, but `.bz2` or `.lzma`, since it makes little sense to use a compression format that is designed to handle multiple files and directory structure to compress what is always a single disk image, and also since, up until very recently, 7-zip and the `.7z` format was mostly geared towards Windows usage, whereas most images are generated on non Windows OSes, where `.7z` would not be the prime candidate for archive compression.

All this to say: I'll take a Bled patch if someone is willing to contribute one. But if not, you'll just have to wait until I either am running out of higher priority things to do, or, the provision of `.7z` compressed images becomes a lot more widespread than it is now.

As such, I'm afraid I will close this request.

pbatard, 2021-03-29:
Btw, I'll add that the Bled code is pretty much derived straight from the methods of compression that `buysbox` added support for, so, if `busybox` didn't really see the need to add support for `7z` compression, we kind of feel justified to follow in their track.

closed 2021-03-29 by pbatard

MichaIng, 2021-03-30:
Fair enough. I agree that there are not many images shipped 7z-compressed. We at DietPi [do](https://dietpi.com/downloads/images/), but I didn't see any other case, and I keep the practice mostly for historical reasons, while `.xz` and `.lzma` would be proper alternatives.

> it makes little sense to use a compression format that is designed to handle multiple files and directory structure

We add readme and hashes into the archive, but a tarball would do as well. Now that Igor from 7-Zip started to release official Linux binaries (opposed to the 3rd party unmaintained `p7zip`), it might be used more in the future, we'll see.

Adopting the WMI solution and using the pre-installed 7-Zip instance on Windows isn't as easy, is it?