pbatard/rufus #1708: A disk read error occurred when booting from USB

Nikita-stepan0v, 2021-03-29:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
A disk read error occurred when booting from USB

Log
---
```
<FULL LOG>
```

pbatard, 2021-03-29:
Please follow the check list and run a bad blocks check on your media. As detailed by the check list, Rufus can perform that bad blocks check for you.

A disk read error usually indicates that your USB drive or the ISO you used have an issue, and it is therefore not something that Rufus can do anything about.

Also, if you encounter a problem when booting media created by Rufus, then your log should show the ISO you used and how you created the media. Without that information, it's next to impossible to help you.

pbatard, 2021-04-03:
No update &rarr; Closing.

closed 2021-04-03 by pbatard