pbatard/rufus #1713: W7 MBR with UEFI:NTFS

marcosfrm, 2021-04-03:
It would be nice have Windows 7 MBR installed with UEFI:NTFS when dual UEFI/BIOS mode is enabled and selected, restricted to MBR partition type of course. This combination (UEFI:NTFS + dual UEFI/BIOS mode) is not allowed today.

If I create an UEFI:NTFS disk in Rufus, copy Windows 10 2009 x64 files to the NTFS partition, then just add the proper MBR code in Linux

```
# ms-sys --mbr7 /dev/sdb
```

will make it work with both UEFI and BIOS, since the NTFS partition is active.

Can be further enhanced if exFAT partition boot record is present (#1631), this way NTFS and exFAT (supported by UEFI:NTFS) would work.

```
Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19042.867)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
Found USB 2.0 device 'Kingston DT 101 G2 USB Device' (0930:6545)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 948, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 0, size: 7.3 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 7.3 GB)
● Creating UEFI:NTFS Partition (offset: 7802097664, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Zeroed MBR
Found volume \\?\Volume{19844501-9469-11eb-9229-7440bbff1e80}\
Disabling file indexing...
Successfully remounted \\?\Volume{19844501-9469-11eb-9229-7440bbff1e80}\ as E:
Finalizing, please wait...

Found USB 2.0 device 'Kingston DT 101 G2 USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 948, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xC805074B
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.3 GB (7801049088 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 15238472, Boot: No
```

pbatard, 2021-04-03:
You may want to try the [<kbd>Alt</kbd>-<kbd>E</kbd> cheat mode](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes).

Unfortunately, you chose to ignore the check list, which points you to the FAQ.

Last time I checked, you could create what you request by using <kbd>Alt</kbd>-<kbd>E</kbd>.

Also, I currently see no point of adding exFAT as a choice of file system for Windows boot media, when I am not aware of any issue with NTFS, whereas exFAT is actually not supported by the Windows 7 boot loader.

As I believe I have answered your question, I will close this issue.

closed 2021-04-03 by pbatard

marcosfrm, 2021-04-03:
Alt+E is enabled here Pete.

pbatard, 2021-04-03:
Then what you request should work.

I just created a media with <kbd>Alt</kbd>-<kbd>E</kbd> enabled, from the 20H2 ISO (which requires UEFI:NTFS for UEFI boot, since it has a > 4 GB `install.wim`) and both the UEFI:NTFS partition was created and the MBR bootloader was added.

```
Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19042.867)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Found USB device 'Kingmax Kingmax USB Device' (1687:0903) [ID]
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3951, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {925B21DB-11B3-421F-9A67-53F850E98E47}
Max parts: 128, Start Offset: 17408, Usable = 32505821696 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'Tails'
  ID: {6C191B28-B30B-434E-9EBA-5FE7224ACDA8}
  Size: 8 GB (8589934592 bytes)
  Start Sector: 2048, Attributes: 0xD000000000000005
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-GB_DV9'
  Size: 5.8 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_20H2_v2_EnglishInternational_x64.iso (5.8 GB)

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 8 GB)
Opened \\.\PhysicalDrive5 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 30.3 GB)
● Creating UEFI:NTFS Partition (offset: 32504774656, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume \\?\Volume{461cce9b-9410-11eb-8c5b-f0038c9a2e6f}\
Disabling file indexing...
Opened \\?\Volume{461cce9b-9410-11eb-8c5b-f0038c9a2e6f} for exclusive write access
Writing Partition Boot Record...
Using Standard NTFS partition boot record
[... 941 lines cut ...]
  25608748 KB available on disk.
      4096 bytes in each allocation unit.
   7935479 total allocation units on disk.
   6402187 allocation units available on disk.
Total duration: 212.47 milliseconds (212 ms).
NTFS Fixup completed.

Found USB device 'Kingmax Kingmax USB Device' (1687:0903) [ID]
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3951, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x06393488
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 30.3 GB (32503726080 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 63485888, Boot: No
```

Can you please detail your issue further, because, as far as I can tell, Rufus already installs both supports BIOS with MBR bootloader + UEFI with UEFI:NTFS bootloader at the same time, and I really don't get why you are adding your own MBR bootloader then.

marcosfrm, 2021-04-03:
I was selecting "UEFI:NTFS" in "Boot selection". In this case, dual mode is not available. I must just select my Windows ISO and set the filesystem to NTFS!

Sorry for the noise.

pbatard, 2021-04-03:
Ah, that makes sense now.

Indeed, if you select the UEFI:NTFS option, you don't get dual boot (but you get exFAT support), as this advanced option is designed for UEFI boot only and no BIOS boot record is installed even if you pick MBR as the partition scheme. But it wouldn't really make sense to install one anyway, as there is no telling if someone is using the standalone UEFI:NTFS option for Windows boot, or for Linux, or for something else (or, more likely, for no BIOS boot whatsoever, as the MBR code is only one part of the BIOS boot process, and there must be a compatible supplementary boot loader in the partition, such as a `bootmgr` in the case of Windows, or a Syslinux or GRUB loader in the case of Linux, and you only get one of those with specific BIOS-compatible ISOs). So, even if we wanted to, it wouldn't make sense to force the Windows MBR to be installed when UEFI:NTFS is selected.

On the other hand, once the use has selected the ISO, then we know if it is BIOS compatible, what type of BIOS bootcode it needs and other things, such as whether it also requires UEFI:NTFS. So it's not that complicated to have Rufus take care of filling all of these automatically, along with the copying of the ISO files.

I'll just remind you that, even if you don't think it really applies, attaching the log helps ensure that we don't waste too much time on back and forth, as it would have been obvious that you only created the UEFI:NTFS partition then, rather than have it done as part of the Windows ISO processing. So please try to stick to the check list as much as possible, thanks.

marcosfrm, 2021-04-03:
> Ah, that makes sense now.
>
> Indeed, if you select the UEFI:NTFS option, you don't get dual boot (but you get exFAT support), as this advanced option is designed for UEFI boot only and no BIOS boot [...]

Yeap. Thanks.

> I'll just remind you that, even if you don't think it really applies, attaching the log helps ensure that we don't waste too much time on back and forth, as it would have been obvious that you only [...]

The log I posted in https://github.com/pbatard/rufus/issues/1713#issue-849687411 is from "UEFI:NTFS" selected in "Boot selection".

pbatard, 2021-04-03:
> The log I posted in #1713 (comment) is from "UEFI:NTFS" selected in "Boot selection".

Shoot, I'm really blind today. My bad.

marcosfrm, 2021-04-04:
Off-topic question: does Rufus patch uefi-ntfs.img before writing it to the 512 KB partition? I am thinking if I could use uefi-ntfs.img on Linux, dd'ing it to a partition of the same size (on the begining of the disk though).

JonnyTech, 2021-04-04:
> Off-topic question: does Rufus patch uefi-ntfs.img before writing it to the 512 KB partition? I am thinking if I could use uefi-ntfs.img on Linux, dd'ing it to a partition of the same size (on the begining of the disk though).

No patching required, `dd`ing is what [WoeUSB](https://github.com/WoeUSB/WoeUSB/blob/master/sbin/woeusb#L1068) does