pbatard/rufus #1714: 2 new usb are not working after using Rufus to format lubuntu

jaimefebe, 2021-04-06:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
First of all, sorry about my English, I am tired and this subject is not my strongest one.
I received today 2 new USB pendrives of the brand Verbatim “PinStripe USB 3.2 Gen 1 Drive”. The device ID is “VID = 18A5 PID = 0243”.
I decided to try the distro Antix (antiX-19.3_x64-base) in one of them, and I used Rufus, but I didn´t like Antix so I decided to try Lubuntu 20.10.
I downloaded Lubuntu´s ISO, selected to create the maximum amount of persistent partition, and I clicked "Start". When Rufus was working I got an error, but I didn´t save that log because I thought It was just a matter to do it again.
I found that it was not possible to read the pen through windows, and the "Disk management" app of windows wasn´t able to detect it.
I decided that later I would format it through linux. 
Then I opened the blister of the other new USB and tried again to put Lubuntu on it.
Again, it appeared the same problems. This time I saved the Log (That is below)

So, I decided to run an ready USB with the gparted live disk (1.2.0.1) on another computer. I ran the application, I plugged the USB I selected “GPARTED” -> “Refresh Device”. It started scanning all devices, but I got the message 

> Could not stat device /dev/sdc – No such file or directory

 And on the list of devices only appears the main USB drive and my HDD. No sign of the damaged USB.  That happens with both USB.

I decided to run an Ubuntu 20.04.2.0 LTS live distro because it has more tools. At Ubuntu Gparted gives me the same error.
I try to check if I am lucky with the command "lsusb". I found that the device is detected for around 50 seconds or similar 

> Bus 002 Device 011: ID 18a5:0243 Verbatim, Ltd Flash Drive (Store'n'Go)

, but it disappears automatically. Every time I plug any of both USB pendrives does the same.

I go to the  "Disks" application, that has a GUI. Then I try again to plug in and plug off bot pendrives. Both appear on the list for around 55 seconds, and then they disappear. If I try to format any of the pendrives through the "Disks" applications it´s no successful because after the infamous 50 seconds a message appear:  

> Error creating partition
> Error creating partition on /dev/sdc: Device'/dev/sdc' invalid or not existing (Could not stat device /dev/sdc - No such file or directory.) (udisks-error-quark, 0)

And this happens using different USB ports of the same laptop, no matter if they are 2.0 or 3.0 

In conclusion: 

- The pendrives are new but they failed after using Rufus. 
- The same operation with Rufus produced the same failure.
- The pendrives are detected by the system for a brief moment.

Please, let me know if I can do more tests that can help you to identify this problem, or if you need more details about the above descriptions.

Log
---
```
Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19042.867)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1912, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0757FC94
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15728517120 bytes)
  Start Sector: 240, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.04
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Lubuntu 20.10 amd64'
  Size: 1.8 GB (Projected)
  Uses: GRUB2
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: lubuntu-20.10-desktop-amd64.2.iso (1.8 GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 122880, size: 14.6 GB)
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has a Windows 2000/XP/2003 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 2.6 GB)
● Creating Linux Persistence Partition (offset: 2843737600, size: 12.0 GB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

0 devices found
0 devices found
0 devices found

```

pbatard, 2021-04-06:
Well, one thing you need to understand that, even as it repartition and reformats a drive, there's simply no way (and you can ask any specialist you want to confirm that) for Rufus to send "special data" to a drive in a manner that will damage it.

It's just the way Windows works: Windows presents a USB drive, __like any other disk device__, as a bunch of sectors that can be written, and, and this is key, there is no special sector on a disk that can render it inoperable.

In other words, as long as there isn't a hardware failure, you can write all the garbage you want to any sector of a disk, and it will never ever render that disk inoperable.

Oh and furthermore, as far as USB access is concerned, when Rufus writes to a disk, it actually doesn't do anything specific with regards to whether it's a disk that is connected through the USB bus, SATA bus, NVMe, or even a VHD. In fact, one of the first thing Rufus does during enumeration, is prevent disks of a certain type from being accessed (such as internal system drives), because Windows presents them in the exact same way, and accessing them is genuinely a matter of telling Windows "write this data at disk sector n".

What this means is that, despite what a lot of people appear to be thinking, Rufus can actually not send "special USB commands" to a flash drive, that would make it inoperable, and even if it were to mess up the partition table or the file system, __as long as there doesn't exist a hardware failure, or a major Windows bug__, no matter how much "bad data" Rufus could send to a drive, that drive will  __always__ be recoverable.

Which means that, the issues you are experiencing have to be caused by the hardware, or by something else.

The first thing that pops to mind is that, since these are 2 new brand drives of the same type, is that bad batches happen. Or (and this is sadly still too common), you may have been sold "fake drive" by your vendor, knowing that a "fake drive" will start to fail once you write something that goes over its actual capacity, which may explain why it only started to fail once you wrote the Ubuntu image. Or it could be bad flash (which is actually not that reliable, and may very well fail early in the life of the drive).

Or, it could be a bad controller firmware from Verbatim, like we have [seen this happen with some PNY drives](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Hardware). Or you could be running into one of a the few bugs that Windows has when it comes to refreshing the partition layout of a drive and make it look like it is accessible (though, if you tested on Linux, we can eliminate that one).

I am therefore afraid that there really isn't much I can advise you to do but RMA these drives, if you cannot recover them in Linux, because they clearly have an issue, which, and I (as well as any other developer that deals with Windows APIs to write data to disk drives) can guarantee that it was caused by something external to Rufus.

For the record, when you are getting a brand new drive, it is usually a good thing to run a bad blocks check (and preferably one that includes "fake drive" detection &mdash; Rufus does have such a check) to validate that the drive you are using is actually fit to use.

And of course, I appreciate that you might have a hard time "swallowing" that Rufus had nothing to do with your issue (because, after all, the developer __must__ be biased when it comes to defending their application). Except that what I am telling you above can be independently verified if you want, from people who have some knowledge of how Windows abstracts access to the hardware, such as disk drive, and how it is not that uncommon to get "dodgy" hardware that fails early, especially when it comes to anything that is flash based.

Or you may also want to read the [dedicate FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive) we have on the subject, or perform a search with regards to reports of "drives damaged by Rufus" (and for good measure, "drives damaged by Etcher" or other applications), as you will find that, despite being literally downloaded millions of times, the number of these reports are about the ones you'd expect from coincidental hardware failures (i.e. failures that would have happened regardless of the application used to write to the drive).

jaimefebe, 2021-04-07:
Hi @pbatard 

Thanks for your quick response. Next time I buy a USB drive I will run a bad blocks test and a fake test to avoid problems.

I just have 1 question: Is there anything I can run at windows or Linux that can help you to identify the problem with my pendrives? Maybe it doesn´t recover my pendrives, but I would like to do something useful with them before throwing them to the bin.

pbatard, 2021-04-07:
> Is there anything I can run at windows or Linux that can help you to identify the problem with my pendrives? 

I'm afraid that would have been running a bad blocks check. If the drives aren't accessible, there really isn't much you can do.

>  before throwing them to the bin.

If you are simply going to recycle them, rather than return them to the manufacturer for exchange, something tells me that whoever sold you those drives isn't a reputable vendor, because if they were, you should be able to return them for exchange. If that is the case, this increases the chances that you were sold "fake drives" by that vendor, which would explain your issue (though I would expect Verbatim PinStripe USB 3.2 Gen 1 flash drives to be fairly new, and therefore make it unlikely that someone has already devised a way to alter their internal firmware to make them report a wrong capacity).

I would however still try to return these drives to the seller, as it's always possible that, being fairly new, those Verbatim drives have an issue with their internal firmware (we've seen something like that with PNY drives) that caused them to fail.

Only the vendor/Verbatim will be able to tell you what happened...

closed 2021-04-07 by pbatard

jaimefebe, 2021-04-07:
Thanks for the suggestion but the reason why I will thrown them is simply because with the pandemic returning them will be a pain in the ass, so I prefer to spend that time researching for a more trustworthy brand to avoid future headaches.