pbatard/rufus #1717: Operation Fails cause of Timeouts - Windows PRO 10 ISO

lmcdasm, 2021-04-15:
﻿

Checklist
---------
- [ X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Tried to burn a ISO with Rufus, falis with timeout.  dispart, fsck and other utils seem to work fine and the disk is working fine.
<FULL LOG>
```

![image](https://user-images.githubusercontent.com/9358276/114939970-23f6b180-9e0f-11eb-88fc-dbcce41893a3.png)

pbatard, 2021-04-15:
Please provide the log __as required per the checklist__.

lmcdasm, 2021-04-15:
Thanks..   I found a work around . Lots of Google reports of the same
issue...

Take care

closed 2021-04-16 by pbatard