pbatard/rufus #1720: Error: ISO image extraction failure...

Tamago4a, 2021-04-24:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm trying to install Kali Linux on an 8GB flash drive. Actually, that's all ...
[image]

Log
---
```
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/20200816/' for Syslinux installation
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 3.4 GB)
● Partition 2 (offset: 3675265024, size: 3.9 GB)
Opened \\.\PhysicalDrive4 for shared write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Bad Blocks: Checking from block 0 to 59615 (1 block = 128 KB)
Bad Blocks: Using offset 97693 for fake device check
Bad Blocks: Writing test pattern 0x55
Closed Windows format prompt
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.4 GB)
● Creating Linux Persistence Partition (offset: 3675265024, size: 3.9 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
252606 possible inodes out of 1010425 blocks (block size = 4096)
50521 blocks (5.0%) reserved for the super user
Creating 31 inode sets: [1 marker = 1.0 set(s)]
+++++++++++++++++++++++++++++++
Creating 8192 journal blocks: [1 marker = 102.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Created 'persistence.conf' file
Done
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
[... 1125 lines cut ...]
Retrying in 5 seconds...
Search for conflicting processes was interrupted due to timeout
Write error [0x00000070] There is not enough space on the disk.
  Error writing file: [0x00000070] There is not enough space on the disk.
Extracting: F:\pool-udeb\main\o\openssl1.0\libcrypto1.0.2-udeb_1.0.2q-2_amd64.udeb (863.4 KB)
  Unable to create file: [0x00000003] The system cannot find the path specified.
Created: F:\syslinux.cfg → /isolinux/isolinux.cfg

Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
Using autorun.inf label for drive F: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x009D0F17
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.4 GB (3674216448 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 3.9 GB (4138702848 bytes)
  Start Sector: 7178252, Boot: No
```

Tamago4a, 2021-04-24:
As soon as I changed the maximum value "3947" MB to "3584" MB, Rufus was able to install the image normally. But I would like to use all the memory...
[image]
Log
---
```
Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
Using autorun.inf label for drive F: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 7.8Гб, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x01017A2F
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.4 GB (3674216448 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 3.9 GB (4138702848 bytes)
  Start Sector: 7178252, Boot: No

Computing checksum for 'E:\Загрузки\kali-linux-2021.1-live-amd64.iso'...
  MD5:    3a3716fef866e5c29a1c1ccfc94264b5
  SHA1:   c92f6766ab9227aa6b546499254b15a1976785f7
  SHA256: 8e5af78e93424336f787d4dd0fdd89b429675d5ae67b1c1634ea1b53c5650677
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/20200816/' for Syslinux installation
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 3.4 Гб)
● Partition 2 (offset: 3675265024, size: 3.9 Гб)
Opened \\.\PhysicalDrive4 for shared write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.8 GB)
● Creating Linux Persistence Partition (offset: 4055885824, size: 3.5 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
229375 possible inodes out of 917500 blocks (block size = 4096)
45875 blocks (5.0%) reserved for the super user
Creating 28 inode sets: [1 marker = 1.0 set(s)]
++++++++++++++++++++++++++++
Creating 8192 journal blocks: [1 marker = 102.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Created 'persistence.conf' file
Done
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{6b4bd711-a51f-11eb-b38d-2cf05d929d71}\
Installing Syslinux 6.04...
Opened \\?\Volume{6b4bd711-a51f-11eb-b38d-2cf05d929d71} for shared write access
Using existing './rufus_files/syslinux-6.04/20200816/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.04/20200816/ldlinux.bss' ✓
[... 1205 lines cut ...]
Extracting: F:\sha256sum.txt (126.6 KB)
Extracting: F:\tools\loadlin.exe (60.5 KB)
Extracting: F:\tools\loadlin.txt (32.0 KB)
Extracting: F:\win32-loader.ini (228 bytes)
Created: F:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...
F:autorun.inf already exists - keeping it

Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
Using autorun.inf label for drive F: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 7.8Гб, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x011C44A4
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.8 GB (4054837248 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 3.5 GB (3758082048 bytes)
  Start Sector: 7921652, Boot: No
```

pbatard, 2021-04-24:
Sorry, but if you set persistence, and want to use the maximum size you can, you need to __manually__ adjust your size until you see the `There is not enough space on the disk.` errors go away.

Rufus can only provide an __estimate__ of how much space an ISO will take when extracted, and there is simply no way we can provide something that's accurate down to a few MB (there are simply too many factors that impact the actual size that will be taken on disk).

As such, if you are floating to close to these limits, and they happen to be too small, it's entirely your job to find out what size you should use for persistence, so that enough space is left for the system.

I will therefore close this issue.

closed 2021-04-24 by pbatard

Tamago4a, 2021-04-25:
It would be nice to somehow warn people about this error in more detail. Because not everyone, like me, is trying to experiment with the parameters and look for the problem on their own.