pbatard/rufus #1726: Sir Virustotal say its a virus

maxmod, 2021-05-05:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Sir,
When we test https://github.com/pbatard/rufus/releases/download/v3.14/rufus-3.14.exe on virus total its good
but when i test my compiled version no changes it say virus please take a look
official rufus
https://www.virustotal.com/gui/file/c0ccf4f480545b50169cc1f5bf92b357ce588520cb8534128200ca48fc6ae588/detection
my compile version no changes
https://www.virustotal.com/gui/file/794ef6d2acb3a9ed3d77a20539ddedda505d1d3316644180c9980169679b1f0a/detection

Log
---
```
<FULL LOG>
```

pbatard, 2021-05-05:
Short version: It is entirely **your** responsibility to demonstrate to Antivirus vendors that an executable **you** compiled does not contain malware. Even if you did not alter the source, I can not vouch for an executable that was produced on a machine and with a toolchain that I have no insight into.

Long version:

1. I am not seeing your issue. [Here](https://www.virustotal.com/gui/file/a608fa527937d7738ce193e7d72184565638b2b1dd5c26405a1cf79b4e2a55c1/detection)'s the VirusTotal report of a Rufus 64-bit executable that I have just compiled on a Windows 10 x64 PC using an up to date Visual Studio 2019 toolchain. At the time of writing this post, VirusTotal reports that _No security vendors flagged this file as malicious_. But please bear in mind that you are really lacking details on how exactly you compiled your version. For a start, you don't indicate the toolchain you used or the architecture you produced, which makes it impossible to try to reproduce and corroborate your claim (and as a matter of fact, when I try to replicate it, I see nothing).

2. At any rate, if some Antivirus vendors report an issue with **your** executable, then it's up to **you** to contact those various Antivirus vendors and report a __false positive__. Especially, you may want to ask them to point out what specific part of the Rufus source is supposed to trigger the malicious behaviour they report. But you also probably want to be able to provide details about the toolchain you used and how exactly you compiled the source, because, as I said above, this may be critical to your problem.

I have to stress out however that the only versions of Rufus I can vouch for are the ones I compile myself, because there is no way I can tell that your environment is set to produce an executable in a safe fashion. The moment you compile your own version, and its behaviour differs from the one I produce (including when it comes to Antivirus reports), you are entirely on your own.

closed 2021-05-05 by pbatard

maxmod, 2021-05-06:
Sir,
64 bit version no issue issue in 32 bit compile it say virus  using visual studio 2019 screenshot of my system config sdk etc
https://prnt.sc/12kdn46
Thank You

pbatard, 2021-05-06:
Well, the fact that the 64-bit version of an executable shows no issue tells you everything you need to know about the reliability of the engines that report malware.

Again, please contact the Antivirus vendors about this, because, as far as I'm concerned, I consider that I've [wasted more than enough time fighting false positives](https://github.com/pbatard/rufus/wiki/FAQ#Antivirus_X_reports_that_Rufus_contains_malware) and I just don't have the time to go around demonstrating, yet again, that those alleged "security experts" are doing a BAD JOB...

blackcrack, 2021-05-06:
[joke]
Master Batard, only you be in possibility to compile the Source with you magic hands, 
by other will only born a 32BBBBBastard as Binarybaby ..
Only youu maaa Master ... have the magic Fingers ...
(schhhhk...schhhhhk)
[/joke]

by the way, a baby can name it self also as Security Expert, it only depends on how much knowledge the person has..
so.. not the Hackers be all bad, only the one with a Black hat.... 
So, Only the good Jedi with knowledge is a good Expert, the others.. can you kick all in the Ton !

best and a nice Day :)
best
Blacky
Blacky

pbatard, 2021-05-06:
@blackcrack, I had to remove your comment as I don't consider it brought anything of value to the discussion. Please keep in mind that this is not a discussion forum, where positing opinions and/or jokes is welcome.

blackcrack, 2021-05-06:
hi @pbatard . is okey Pete, i hope you had laughed ;)
see u :)
best