pbatard/rufus #1728: [bug] Windows 3.14.p causes dGPU to wake up for no reason on start and exit

makedir, 2021-05-07:
Starting Rufus 3.14p on Windows 10 on a dual GPU laptop with Intel and Nvidia GPU, causes the dGPU to wake up for a few seconds starting Rufus for no reason. Waiting until it shuts down and then closing Rufus, after 3-4 seconds of closing Rufus, the dGPU wakes up again for no reason for a few seconds.

makedir, 2021-05-07:
check.... check.... and the log wont help you anything with this bug report, but sure:

Rufus x86 v3.14.1788 (Portable)
Windows version: Windows 10 64-bit (Build 19042.964)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
0 devices found

pbatard, 2021-05-07:
What does the Windows Event log say? It might be possible to query the reasons why you dGPU wakes up.

I would also like to understand how exactly this is an issue. I doubt the OS can provide you with any guarantees that a GPU will not be woken up at random, so I don't really get how you believe that someone should devote resources ensuring that what looks like a very minor problem, that has no impact on your ability to carry out your regular operations, should not occur. What exactly is the downside of having your dGPU wake up for a few seconds?

Now, obviously, there's nothing in the Rufus code even remotely related to GPU or 3D access, but we do however need to perform some generic operations to get the UI to display properly. So my guess is that the Windows __standard__ UI APIs may cause GPU wakeup. But I can't say I am planning to go out of my way to find out what specific API is responsible for this, as the end result will be me telling you that you will need to contact Microsoft if you have an issue with this behaviour, because there really is no code in Rufus that relates to 3D/GPU access...

pbatard, 2021-05-07:
> I would like to find out why this happens because I have rare dGPU wake ups in the past in Windows 10, never found out why it happens.

Sorry, but if this is the extent of your problem, I really can't see myself investing time to help you figure that out, because, as far as I am concerned, it has no impact on users, except a handful of people who appear to get annoyed that some components wake up.

I'm afraid that all I can say here is: You have access to the source and you can easily compile and debug your own version of Rufus, so, if it is that important to you, then I will respectfully ask that you invest your own time investigating why this very minor issue happens (and possibly report your findings here).

However, my time being in too short supply to invest it on low priority issues such as this one, I'm just going to have to close it for now. Yes, I do understand that you are annoyed that Rufus wakes up your GPU. But no, I am not going to go out of my way to figure out the cause when I have issues and features to address, that will benefit a much larger number of people. Sorry.

closed 2021-05-07 by pbatard

makedir, 2021-05-07:
Can you please dont be like that. I didnt want to sound rude if I did I apologize. There has to be something special in Rufu causing this. Look at this video

You can see the event happening in this video:

https://www.dropbox.com/s/tknwrccx9k5mwsr/2021-05-07-1529-26.flv?dl=1

Who else can help me with this then you? I am not a c programmer I dont have knowledge about that.

There is some routine in Rufus which accesses the dGPU for whatever reason.

pbatard, 2021-05-07:
You do realize that you are asking me to do work, __for free__, for what I can only qualify as a minor inconvenience.

Well, it's not because I do provide the Rufus application for free that my time is also free (or even that I am selfless and will attempt to solve every user's problem).

Investigating your issue does have a price because I do consider my time as valuable. And I don't see why I should invest my valuable time, for free, for something that has no real impact on users.

So, again, I will kindly request that people who are interested in figuring out the cause of this issue (of which I am not) do invest their own time, if they see it as that important.