pbatard/rufus #1731: My Samung Portable SSD T5 did not get recognised?

LowLevelMahn, 2021-05-08:
Version: Rufus 3.14

i've got 4 USB-Sticks and a 

Samung Portable SSD T5
https://www.samsung.com/de/memory-storage/portable-ssd/portable-ssd-t5-2tb-black-mu-pa2t0b-eu/

Rufus showing me all my USB Sticks but not the Portable SSD, does Rufus only work with "real" USB Sticks?

pbatard, 2021-05-08:
Closed on account of not following the check list. Following the check list would have given you no less than **THREE** different ways to answer your question, and I am growing exceedingly tired of people who believe they are above following the checklist whereas they are the __very ones__ the check list is here for.

closed 2021-05-08 by pbatard

LowLevelMahn, 2021-05-08:
So sorry, found the answer in your really well prepared FAQ list