pbatard/rufus #1732: The appx for 3.14 is not available in rufus.ie/downloads

superusercode, 2021-05-10:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Despite 3.14 being available in the windows store, it seems to have missed the downloads page. Is there any way we can get that before 3.15?

Log
---
```
<FULL LOG>
```

pbatard, 2021-05-10:
> Despite 3.14 being available in the windows store, it seems to have missed the downloads page.

No, that is intentional. The `.appx` was only provided because Rufus was not available from the Windows Store. But now that we've finally managed to get Rufus in the Windows Store, providing the `.appx` is completely superfluous, so I won't be providing one any longer.

If you want Rufus as an app, get it from the Windows Store.

closed 2021-05-10 by pbatard

JonnyTech, 2021-05-11:
As I remember, the `.appx` version contained an uncompressed (non UPX'd) build, is this available elsewhere other than the Windows store?

pbatard, 2021-05-11:
Last time I checked, if you know what you're doing, you can download the Windows Store package bundle, open it in 7-zip, and get the uncompressed executables from that bundle.