pbatard/rufus #1733: Rufus not resizable so Os not appearing

shadow-hunte, 2021-05-14:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I have a windows AIO with 140 windows version and i'm trying to create a windows to Go Bootable usb but Rufus is not listing all the windows version and i'm unable to increase the size of the windows because all windows in rufus are not resizable. I believe all are showing but because the window size is too small, i cant see everything>

Log
---
```
<FULL LOG>
```

JonnyTech, 2021-05-14:
> I have a windows AIO with 140 windows version

I that a legal and official version??

pbatard, 2021-05-14:
Windows To Go is only supported for __official__ Windows retail ISOs. You are not using one of those.

If you can find an official ISO (such as the ones downloaded from https://www.microsoft.com/en-gb/software-download/windows10ISO) that exhibit your problem, I will look at it. But until then, I will just close this issue.

closed 2021-05-14 by pbatard