pbatard/rufus #1734: Copycat / possible GPL violation

dreua, 2021-05-19:
Github user "maxmod" advertised a software, probably theirs, here: https://github.com/FedoraQt/MediaWriter/issues/216#issuecomment-843822277

The link led to https://formatusb [dot] com/download/ (not linking on purpose, proceede with care, the whole thing looks sketchy to me).

The screenshots there look very much like rufus.

Just wanted to make you aware of this, maybe you want and can (via license enforcement) do something about that.

pbatard, 2021-05-19:
Yup, definitely a copyright violation.

Anybody can create a derivative or a copy of Rufus if they wish, but if they do so, they **MUST** publish the source code of their work, and these guys aren't. So they are in breach of the GPL.

I'll see what I can do. Thanks for the report.

usbtool, 2021-05-20:
> Yup, definitely a copyright violation.
> 
> Anybody can create a derivative or a copy of Rufus if they wish, but if they do so, they **MUST** publish the source code of their work, and these guys aren't. So they are in breach of the GPL.
> 
> I'll see what I can do. Thanks for the report.

Sir,
We are optimizing the code so delay now souce code published new to github
Thank You

usbtool, 2021-05-20:
> Github user "maxmod" advertised a software, probably theirs, here: [FedoraQt/MediaWriter#216 (comment)](https://github.com/FedoraQt/MediaWriter/issues/216#issuecomment-843822277)
> 
> The link led to https://formatusb [dot] com/download/ (not linking on purpose, proceede with care, the whole thing looks sketchy to me).
> 
> The screenshots there look very much like rufus.
> 
> Just wanted to make you aware of this, maybe you want and can (via license enforcement) do something about that.

Sir,
We are optimizing the code so delay now you can download the souce code
Thank You

usbtool, 2021-05-20:
> Yup, definitely a copyright violation.
> 
> Anybody can create a derivative or a copy of Rufus if they wish, but if they do so, they **MUST** publish the source code of their work, and these guys aren't. So they are in breach of the GPL.
> 
> I'll see what I can do. Thanks for the report.

Sir,
Source code on https://github.com/usbtool/formatusb if still violatin please guide me how to remove
Thank You

usbtool, 2021-05-20:
> Yup, definitely a copyright violation.
> 
> Anybody can create a derivative or a copy of Rufus if they wish, but if they do so, they **MUST** publish the source code of their work, and these guys aren't. So they are in breach of the GPL.
> 
> I'll see what I can do. Thanks for the report.

Sir,
Big Fan of Rufus from verion 1 release of rufus please guide what i made wrong
Clearly mention 
This software is based on Rufus https://rufus.ie/en/     https://github.com/pbatard/rufus
On main page also

dreua, 2021-05-20:
If I may add two points:
- Do not post bit.ly links to your software (don't use shortened URLs at all here on Github, there is no point in doing so)
- Do state your authorship of FormatUsb if you post links or recommend it.

usbtool, 2021-05-20:
> * _Temporarily_ delete your `formatusb` GitHub repository.
> * Fork Rufus, from the commit you used for your base, as a new `formatusb` GitHub project.
> * Apply your changes on top of that fork and whatever else you want to provide such as binary releases.

Sir,
Thank you for your suggestion
I am new developer and really don't know much about policy etc also github is new for me 
1 step deleted my repo and 
2 fork the rufus
3 uploaded the file using github desktop
4 Mention the rufus on main page of web 
if done anything wrong i am sorry 
Thank You

usbtool, 2021-05-20:
> If I may add two points:
> 
> * Do not post bit.ly links to your software (don't use shortened URLs at all here on Github, there is no point in doing so)
> * Do state your authorship of FormatUsb if you post links or recommend it.

ok

pbatard, 2021-05-20:
This is getting closer to fixing the issues, so thanks for looking into this and applying some of the requested changes.

However, there are still changes for which you have not published the source.

For instance, you did modify the UI to remove the advanced options toggle and you also removed buttons from the 4-icon panel (as well as changed the "information" icon for that panel):

[image]

None of these code alterations appear in your commits at https://github.com/usbtool/formatusb/commits/master so, at this moment, you are still in violation of the GPL, because you did release a GPL licensed binary for which not all of the source code has been made available (and yes, even UI changes are considered to be part of the source). Most likely, you have applied changes to `src/ui.c` and `src/ui.h` as well as `res\icons\` that are not published in your repository yet, so you still have to fix that to be GPL compliant.

If you rectify this, and make sure that the code you have in  https://github.com/usbtool/formatusb allows someone to produce a binary that looks and behave identically to the one you publish (which means that, if I use the code from your repo, then I should be able to get an application that looks exactly like the screenshot above), then everything should be okay.

But I'm afraid that, if you don't want to be in violation of the GPL, you must ensure that you publish every single change you have applied to the original source, no matter how small. A partial set of changes will not do and can still leave you in legal troubles...

At this stage, and considering that you have been eager to follow the earlier suggestions, I'm going to take you at your word that you do want to fix those issues, so if you don't have to rush to push commits if you want to clean up your code before publishing the missing changes. But I would ask you to try to publish commits for all the residual changes you applied within a couple of weeks, if you want to make sure that you can't be pursued for GPL violation, since anybody, and not just the copyright holder of the code you produced a derivative of, can cite you for GPL violation if they can't find the code for all the changes you applied.

If you are starting with Open Source development, I can understand that following the GPL to the letter may sound annoying at first. But as you go along, you'll understand that it's actually how the GPL ensures that software freedoms are preserved and that anyone, like you, who wants to create a derivative of a project can work on a level playing field (which is actually why you were able to create your derivative works in the first place, because the GPL ensured that I could not hold back any code that would have prevented you from creating a 1:1 copy of the Rufus application).

usbtool, 2021-05-21:
> This is getting closer to fixing the issues, so thanks for looking into this and applying some of the requested changes.
> 
> However, there are still changes for which you have not published the source.
> 
> For instance, you did modify the UI to remove the advanced options toggle and you also removed buttons from the 4-icon panel (as well as changed the "information" icon for that panel):
> 
> [image]
> 
> None of these code alterations appear in your commits at https://github.com/usbtool/formatusb/commits/master so, at this moment, you are still in violation of the GPL, because you did release a GPL licensed binary for which not all of the source code has been made available (and yes, even UI changes are considered to be part of the source). Most likely, you have applied changes to `src/ui.c` and `src/ui.h` as well as `res/icons/` that are not published in your repository yet, so you still have to fix that to be GPL compliant.
> 
> If you rectify this, and make sure that the code you have in https://github.com/usbtool/formatusb allows someone to produce a binary that looks and behaves identically to the one you publish (which means that, if I use the code from your repo, then I should be able to get an application that looks exactly like the screenshot above), then everything should be okay.
> 
> But I'm afraid that, if you don't want to be in violation of the GPL, you must ensure that you publish every single change you have applied to the original source, no matter how small. A partial set of changes will not do and can still leave you in legal troubles...
> 
> At this stage, and considering that you have been eager to follow the earlier suggestions, I'm going to take you at your word that you do want to fix those issues, so you don't have to rush to push commits if you want to clean up your code before publishing the missing changes. But I would ask you to try to publish commits for all the residual changes you applied within a couple of weeks, if you want to make sure that you can't be pursued for GPL violation, since anybody, and not just the copyright holder of the code you produced a derivative of, can cite you for GPL violation if they can't find the code for all the changes you applied.
> 
> If you are starting with Open Source development, I can understand that following the GPL to the letter may sound annoying at first. But as you go along, you'll understand that it's actually how the GPL ensures that software freedoms are preserved and that anyone, like you, who wants to create a derivative of a project can work on a level playing field (which is actually why you were able to create your derivative works in the first place, because the GPL ensured that I could not hold back any code that would have prevented you from creating a 1:1 copy of the Rufus application).

Sir,
Yesterday read all gpl copyright instruction and your suggestions  
Same code uploaded you can compile same ui made changes in rufus.rc visibility to hide 
made changes to these files 
resource.h
rufus.c
rufus.h
rufus.rc
stdlg.c
ui.c
ui.h
and icon folders
binary which is on web is same i am using same code to produce it.
Uses Visual Studio 2019 and uploaded via  GitHub Desktop same code uploaded to verify it i requested you to comile it 

Thank You

pbatard, 2021-05-21:
Ah, my bad, it looks like I missed the code changes you had in usbtool/formatusb@3bbba7cfe01868460dab55e04108001d6b7a0703.

There are a few small things that are questionable (such as a [this](https://github.com/usbtool/formatusb/commit/3bbba7cfe01868460dab55e04108001d6b7a0703#diff-0f921eda741b33546cece2f7869409c946c019c3b0dc5b065f1cff7fab614a00L160) where you are replacing the URL for a change where your URL did not apply), which I gather comes from a blanket search and replace of `rufus.ie`, and you have also reverted some of the recent changes I committed, but these are not GPL violation, so, as far as I can see, in terms of licensing you should be okay and I see no major legal problem with your repository as it stands.

I will therefore close this issues. Thank you for acting quickly on resolving the GPL violation issue.

closed 2021-05-21 by pbatard

usbtool, 2021-05-21:
> Ah, my bad, it looks like I missed the code changes you had in [usbtool/formatusb@3bbba7c](https://github.com/usbtool/formatusb/commit/3bbba7cfe01868460dab55e04108001d6b7a0703).
> 
> There are a few small things that are questionable (such as a [this](https://github.com/usbtool/formatusb/commit/3bbba7cfe01868460dab55e04108001d6b7a0703#diff-0f921eda741b33546cece2f7869409c946c019c3b0dc5b065f1cff7fab614a00L160) where you are replacing the URL for a change where your URL did not apply), which I gather comes from a blanket search and replace of `rufus.ie`, and you have also reverted some of the recent changes I committed, but these are not GPL violation, so, as far as I can see, in terms of licensing you should be okay and I see no major legal problem with your repository as it stands.
> 
> I will therefore close this issue. Thank you for acting quickly on resolving the GPL violation issue.

Thank You So Much Sir