pbatard/rufus #1735: Question: Is Microsoft Store version of Rufus is official?

xmha97, 2021-05-22:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [X] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Which version of Rufus do you recommend?
[image]
https://github.com/pbatard/rufus/releases/download/v3.13/Rufus-3.13.appx
https://www.microsoft.com/store/productId/9PC3H3V7Q9CH
Is Microsoft Store version of Rufus is official?

Log
---
```
<FULL LOG>
```

pbatard, 2021-05-22:
> Is Microsoft Store version of Rufus is official?

https://www.microsoft.com/p/rufus/9pc3h3v7q9ch is an official version. I'm publishing it myself, and this is the reason why I have stopped uploading `.appx` packages after 3.13.

> Which version of Rufus do you recommend?

Latest, always. Be it from the store or from the website.

closed 2021-05-22 by pbatard

xmha97, 2021-05-22:
I understand from your description that from now on, you will not upload the `.AppX` version in the Git and the site and should only download from the Microsoft Store. Is it right?

pbatard, 2021-05-22:
Yes that is correct.