pbatard/rufus #1741: Why it takes 10 hours to make a Windows 10 21H1 bootable disk

WinkeyCheng, 2021-05-28:
[image]

WinkeyCheng, 2021-05-28:
Im so annoying :/
I have already wait for 10 hours but only still worked for 42.7%

pbatard, 2021-05-28:
https://github.com/pbatard/rufus/wiki/FAQ#Why_is_the_Windows_To_Go_media_creation_process_so_slow

If you use a drive with slow random I/O writes (doesn't matter if you have fast sequential writes, it's random I/O that's important, and a lot of flash drives are dreadful in that respect), then both the drive creation and your Windows To Go experience will be miserable.

If you want a proper Windows To Go experience, you should use:
1. A USB 3.x SSD enclosure
2. A fast enough SSD (since, unlike flash drives, SSDs are designed for random I/O)

closed 2021-05-28 by pbatard