pbatard/rufus #1745: Using fu

LocTran016, 2021-05-30:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I ran into a problem while I was trying to make a bootable USB to install Window 10 20H1 in Window 7 32bit Starter (VMWare one MacOS) but it ran to a problem here:
[image]
[rufus.log](https://github.com/pbatard/rufus/files/6566263/rufus.log)

pbatard, 2021-05-30:
`Unable to locate FormatEx() in fmifs.dll: [0x0000007F] The specified procedure could not be found.`

I'm afraid your installation of Windows is crippled, as Rufus relies on `fmifs.dll` to work, and yours is either missing or lacking the relevant API calls.

You can try <kbd>Alt</kbd>-<kbd>V</kbd> in Rufus to force the use of VDS for formatting, but I'm afraid I have no plans to try to support versions of Windows where `fmifs.dll` doesn't work as expected, even more so if it's Windows 7 (which has been obsoleted by Microsoft by more than a year now).

As such I will close this issue.

closed 2021-05-30 by pbatard

LocTran016, 2021-05-30:
> `Unable to locate FormatEx() in fmifs.dll: [0x0000007F] The specified procedure could not be found.`
> 
> I'm afraid your installation of Windows is crippled, as Rufus relies on `fmifs.dll` to work, and yours is either missing or lacking the relevant API calls.
> 
> You can try Alt-V in Rufus to force the use of VDS for formatting, but I'm afraid I have no plans to try to support versions of Windows where `fmifs.dll` doesn't work as expected, even more so if it's Windows 7 (which has been obsoleted by Microsoft by more than a year now).
> 
> As such I will close this issue.

I download that file from Microsoft, the Window 10 21H1 ISO but may be I should check again, thanks for your advice

pbatard, 2021-05-30:
> I download that file from Microsoft, the Window 10 21H1 ISO

This is irrelevant. I was talking about the OS that you use to run Rufus on, which is Windows 7 Starter edition running inside a Virtual Machine on MacOS. The ISO you are trying to use has no impact on that.

If you were running Rufus on Windows 8 or 10 and on real hardware, that would be a different story. But you are not, and there's only so much support I can provide for the type of configuration you are using to run Rufus.