pbatard/rufus #1746: garuda not flashing

Martinligabue, 2021-06-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I tried flashing garuda iso with multiple versions and in multiple pcs, but no way to make it work. It creates a 4mb FAT partition (MISO_EFI) in the usb after 2.23gb not allocated and leaving 12.8gb not allocated after. It does not boot, and it contains 132kb of a file "bootx64.efi" in efi/boot/
I cannot try different options because "Note: DD image mode enforced since this ISOHybrid is not ISO mode compatible." (but in garuda website it says to use dd, so there should be no problem), and all options are greyed out.
(PS. I usually use Rufus for everything and it's the only iso it has problems with)

Log
---
```
<FULL LOG>
```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19043.1023)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0410 (it-IT)
Will use default UI locale 0x0410
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'it-IT'
Found UAS (USB 3.0) device 'Seagate Backup+ Hub BK SCSI Disk Device' (0BC2:AB38)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 13142028, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.04
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GARUDA_XFCE_SOARING_'
  Size: 2.2 GB (Projected)
  Uses: GRUB2
  Uses: EFI
Note: DD image mode enforced since this ISOHybrid is not ISO mode compatible.
Using image: garuda-xfce-linux-lts-210507.iso (2.2 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 6728718336, size: 4 MB)
Opened \\.\PhysicalDrive3 for exclusive write access
Requesting logical volume handle...
Bad Blocks: Checking from block 0 to 29327 (1 block = 512 KB)
Bad Blocks: Using offset 112797 for fake device check
Bad Blocks: Writing test pattern 0x55
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++Remounted \\?\Volume{a16ed082-bbeb-11eb-839f-806e6f6e6963}\ as F:

Found UAS (USB 3.0) device 'Seagate Backup+ Hub BK SCSI Disk Device' (0BC2:AB38)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 4678696, Boot: No

Martinligabue, 2021-06-01:
I'm sorry, I found out how to generate the checksum, Ill add it here, if it helps, but I downloaded multiple versions and on multiple pcs, and the problem was always the same

Computing checksum for 'E:\distro\garuda-xfce-linux-lts-210507.iso'...
  MD5:    3ae6b0333d9fb1ee26c57a7f84ff0c26
  SHA1:   c469df47dd996827a678953bbae3921adaaec181
  SHA256: 938e00c6cfac8c411e542c157a07cbfbc879a1cba1fa7eb15ebb7561e564adb1

pbatard, 2021-06-01:
> It creates a 4mb FAT partition (MISO_EFI) in the usb after 2.23gb not allocated and leaving 12.8gb not allocated after.

That's normal. And that 12.8 GB are allocated. They're just not something Windows can mount because it uses a file system that Windows does not support.

Your drive was created __EXACTLY AS EXPECTED__ for DD mode, and if you use Etcher or any other DD imager, you will get the same results.

> It does not boot,

Then you need to take this to the Garuda folks because, contrary to what you seem to believe, I see nothing in your report to indicate that Rufus did not create the drive properly, and again, I strongly encourage you to try with Etcher or even pure `dd` in Linux, where you should find that the result is the same as when the drive is created by Rufus.

Thus, unless you can demonstrate that other utilities manage to create a bootable drive, when writing the Garuda ISO in DD mode, whereas Rufus can, I see nothing to address here, and I will close this issue.

If however, you can show that Rufus differs from other utilities, and that your drive boots with them, please let me know (you can still comment after the issue is closed) and I will reopen this issue.

closed 2021-06-01 by pbatard

Martinligabue, 2021-06-01:
I tried also with etcher, but since I never used that tool I was not sure it wasn't a program fault and it was a while ago so I forgot. I'll try contacting garuda team, now that I think of it I should have gone there first, thanks for now