pbatard/rufus #1747: Error while accessing ldlinux.sys

flyingqube-dev, 2021-06-03:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Help, there is something wrong with accessing ldlinux.sys

Log
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
Removing F: from the list: This is the disk from which Rufus is running!
1 device found
Disk type: Removable, Disk size: 7.7GB, Sector size: 512 bytes
Cylinders: 940, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0066929B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7735023616 bytes)
  Start Sector: 2048, Boot: Yes
Downloading https://rufus.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/20200816/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
Requesting disk access...
Will use 'I:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 7.2 GB)
Opened \\.\PhysicalDrive4 for exclusive write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 7.2 GB)
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{a3d7f6fc-c3a8-11eb-a134-dc1ba1921f4d}\
Installing Syslinux 6.04...
Opened \\?\Volume{a3d7f6fc-c3a8-11eb-a134-dc1ba1921f4d} for shared write access
Could not open rufus_files/syslinux-6.04/20200816/ldlinux.sys
Re-mounted volume as I: after error

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
Removing F: from the list: This is the disk from which Rufus is running!
1 device found
Disk type: Removable, Disk size: 7.7GB, Sector size: 512 bytes
Cylinders: 940, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00680ECB
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)

<FULL LOG>
```

pbatard, 2021-06-03:
Nope, nothing wrong. Especially:

```
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
```
and
```
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓
```
indicate that the files were successfully downloaded.

Please don't mistake expected 404 HTTP codes, that have to do with a fallback mechanism that allows us, __in very specific cases__, to override the generic Syslinux binary with a custom one, with actual errors.

The 404 you see are normal and in no way indicative of an issue. The files you need were also properly downloaded.

As such, since I don't see any evidence of a problem, I will close this issue.

closed 2021-06-03 by pbatard

flyingqube-dev, 2021-06-03:
I meant
`
Could not open rufus_files/syslinux-6.04/20200816/ldlinux.sys`
`Re-mounted volume as I: after error`

flyingqube-dev, 2021-06-03:
[image]
[image]

pbatard, 2021-06-03:
Ah, I missed that.

Well, from what I can see the file you need should have been properly downloaded.

You did truncate the log though, which would have shown important information about the Syslinux version detected.

Can you re-create the media and make sure you copy paste the __full__ log, so that I have all the data needed to properly troubleshoot this issue? Especially, you did check the box that states that you copy/pasted the full log, which is not the case...

flyingqube-dev, 2021-06-03:
[rufus.log](https://github.com/pbatard/rufus/files/6594193/rufus.log)

pbatard, 2021-06-03:
That is __NOT__ the full log.

A full log starts with something like:
```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19043.985)
...
```

flyingqube-dev, 2021-06-03:
but it's all i have

flyingqube-dev, 2021-06-03:
alt+L

flyingqube-dev, 2021-06-03:
wait
i have one

pbatard, 2021-06-03:
Please read what I said. You need to recreate your media and then copy/paste the full log.
The log does start with `Rufus version: x.y.z` always. If you don't see that, you cleared or truncated it for whatever reason, and need to recreate your media. If you want help, it is crucial that I see the log data that you removed.

flyingqube-dev, 2021-06-03:
[rufus-proper log](https://github.com/pbatard/rufus/files/6594212/rufus-proper.log)

flyingqube-dev, 2021-06-03:
sorry

pbatard, 2021-06-03:
Aha:

```
Detected Syslinux version: 6.04/20200816 (from '/isolinux/isolinux.bin')
```

So Kali (why is it always them?) decided to start using slash in their version string. I suspect that this is the cause of the issue. I would really expect Kali folks to know better than use path characters for versioning, but I guess I'll just have to add more sanitizing.

Don't expect anything fast though.

flyingqube-dev, 2021-06-03:
Sure, thanks for patience.

pbatard, 2021-06-03:
Actually, that might not be it, as I tested an older Kali, that uses `Syslinux version: 6.03/20170128` and saw no issue. And I'm pretty sure I already added sanitization.

Especially, I got:

```
Using existing './rufus_files/syslinux-6.03/20170128/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20170128/ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
```

Which means that there's no reason for your `rufus_files/syslinux-6.04/20200816/...` not to have been picked up.

Here is my log:
```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19043.985)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20170128 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 2.7 GB (Projected)
  Note: File on disk is larger than reported ISO size by 704 KB...
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: kali-linux-rolling-amd64.iso (2.6 GB)
Found USB 3.0 device 'Mushkin Ventura Ultra USB Device' (174C:55AA)
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {83766410-FB89-4B6A-B22A-25D2DE6E79C6}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {0401FC27-F41D-4D95-B7F9-0680D49F7F96}
  Size: 111.8 GB (120033058304 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Downloading https://rufus.ie/files/syslinux-6.03/20170128/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.03/ldlinux.sys
File length: 67.0 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.03/20170128/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.03/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
[... 91 lines cut ...]
Extracting: F:\boot\grub\i386-efi\cpuid.mod (1.7 KB)
Extracting: F:\boot\grub\i386-efi\crc64.mod (1.6 KB)
Extracting: F:\boot\grub\i386-efi\crypto.lst (936 bytes)
Extracting: F:\boot\grub\i386-efi\crypto.mod (5.0 KB)
Extracting: F:\boot\grub\i386-efi\cryptodisk.mod (10.1 KB)
Extracting: F:\boot\grub\i386-efi\cs5536.mod (3.8 KB)
Extracting: F:\boot\grub\i386-efi\ctz_test.mod (1.8 KB)
Extracting: F:\boot\grub\i386-efi\date.mod (2.2 KB)
Extracting: F:\boot\grub\i386-efi\datehook.mod (1.8 KB)
Extracting: F:\boot\grub\i386-efi\datetime.mod (1.2 KB)
Extracting: F:\boot\grub\i386-efi\disk.mod (2.4 KB)
Extracting: F:\boot\grub\i386-efi\diskfilter.mod (10.7 KB)
Extracting: F:\boot\grub\i386-efi\div.mod (1012 bytes)
Extracting: F:\boot\grub\i386-efi\div_test.mod (5.5 KB)
Extracting: F:\boot\grub\i386-efi\dm_nv.mod (1.8 KB)
Extracting: F:\boot\grub\i386-efi\echo.mod (1.9 KB)
Extracting: F:\boot\grub\i386-efi\efifwsetup.mod (1.5 KB)
Extracting: F:\boot\grub\i386-efi\efinet.mod (6 KB)
Extracting: F:\boot\grub\i386-efi\efi_gop.mod (7.8 KB)
Extracting: F:\boot\grub\i386-efi\efi_uga.mod (4.5 KB)
(...)
```

Can you please try <kbd>Alt</kbd>-<kbd>D</kbd> in Rufus, to delete the `rufus-files` directory and try again.
I'm not seeing much of a reason for your files not to be picked up...

pbatard, 2021-06-03:
Also, if that doesn't work, can you share a screenshot from file explorer of the content of `rufus_files\syslinux-6.04\20200816\`?

flyingqube-dev, 2021-06-03:
[image]

pbatard, 2021-06-03:
So you are running Rufus from the root of your `F:` drive. What type of drive is it? Is it a physical drive/partition? Something else?

What happens if you try running Rufus from a folder of your `C:` drive. For instance, you can copy the Rufus executable to your desktop and see if it works better from there...

flyingqube-dev, 2021-06-03:
Well, F drive is my pendrive. I already checked and used both portable and normal versions, but both have the same problem. I tried to run it from another druve, ssd, but nothing changed

flyingqube-dev, 2021-06-03:
With c drive... wait until my morning (UTC+2)

flyingqube-dev, 2021-06-04:
Ok, i ran it from C: and now it works.

flyingqube-dev, 2021-06-04:
Maybe it was something with permissions?

flyingqube-dev, 2021-06-04:
[image]

pbatard, 2021-06-04:
Actually, this is a general issue when Rufus is running from the root of a drive, as I was able to replicate the problem on my machine.

From what I can see, the problem is that, in the Microsoft world, something like `chdir("F:\\")`works, whereas `chdir("F:")` doesn't, and in Rufus 3.14, we issue a [`_wchdir()`](https://docs.microsoft.com/en-us/cpp/c-runtime-library/reference/chdir-wchdir) to the application directory, so that we can then reference the `rufus_files\` content.

And of course, in case our application resides on top of the drive, the API we use to locate the application directory ([`GetCurrentDirectory()`](https://docs.microsoft.com/en-us/windows/win32/api/winbase/nf-winbase-getcurrentdirectory)) returns `"F:"` rather than `"F:\\", which makes `chdir()` fail. But as soon as Rufus resides inside a subdirectory of the drive, `chdir()` works.

Now the thing is that, with the commit that I pushed a few days before this issue was opened (9d7e96e293955b27b8424ce86a4490701484895b), this issue has actually become moot because Rufus 3.15 will stop placing downloads into a `rufus_files\` folder residing at the same level as the executable, to instead download into a `Rufus\` directory in `<user>\AppData\Local\` (in which case `chdir()` will work regardless of where the application resides).

But I think I'm going to make sure that I add a `\\` to the application directory string, if it's just a drive letter, just in case Microsoft has an other bad surprise in store with regards to how the data returned by one Windows API is not compatible with another Windows API...

Srv-02, 2021-08-18:
> Now the thing is that, with the commit that I pushed a few days before this issue was opened (9d7e96e293955b27b8424ce86a4490701484895b), this issue has actually become moot because Rufus 3.15 will [...]

Could you also please update the "Usage Notes" regarding "Where to manually copy the Syslinux/GRUB files, if Rufus can't download them"?
I was quite desperate that it didn't work anymore with v3.15 😁

pbatard, 2021-08-18:
> Could you also please update the "Usage Notes" regarding "Where to manually copy the Syslinux/GRUB files, if Rufus can't download them"?

Good point. I'll do that.