pbatard/rufus #1748: Unable to create file - Debian 10.9.0 Bootstick

LeoLTM, 2021-06-06:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I tried to create a Debian 10.9.0 bootstick for my other machine but it crashes with 

Unable to create file: Windows error code 0x000001B1 (FormatMessage error code 0x00003AFC)
Unable to create F:\syslinux.cfg - booting from USB will not work

at 99.2%. 

Log
---
```
Rufus x86 v3.14.1788 (Portable)
Windows version: Windows 10 64-bit (Build 19043.1023)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 3.0 device 'Verbatim STORE N GO USB Device' (18A5:0252)
1 device found
No volume information for drive 0x84
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7703, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x0065ED04
Drive has a Syslinux Master Boot Record
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20190226 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'd-live 10.9.0 gn amd64'
  Size: 2.5 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: debian-live-10.9.0-amd64-gnome.iso (2.5 GB)
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 Byte
Successfully downloaded 'ldlinux.bss'
[... 821 lines cut ...]
Extracting: F:\pool\main\l\lowmem\lowmemcheck_1.47_amd64.udeb (5.1 KB)
Extracting: F:\pool\main\l\lowmem\lowmem_1.47_all.udeb (1.6 KB)
Extracting: F:\pool\main\l\ltsp\ltsp-client-builder_5.18.12-3_all.udeb (9.1 KB)
Extracting: F:\pool\main\l\lvm2\dmeventd_1.02.155-3_amd64.deb (64.6 KB)
Extracting: F:\pool\main\l\lvm2\dmsetup_1.02.155-3_amd64.deb (88.7 KB)
Extracting: F:\pool\main\l\lvm2\dmsetup-udeb_1.02.155-3_amd64.udeb (41.3 KB)
Extracting: F:\pool\main\l\lvm2\libdevmapper1.02.1-udeb_1.02.155-3_amd64.udeb (113.4 KB)
Extracting: F:\pool\main\l\lvm2\libdevmapper-event1.02.1_1.02.155-3_amd64.deb (22.1 KB)
Extracting: F:\pool\main\l\lvm2\liblvm2cmd2.03_2.03.02-3_amd64.deb (625.2 KB)
Extracting: F:\pool\main\l\lvm2\lvm2_2.03.02-3_amd64.deb (1.1 MB)
Extracting: F:\pool\main\l\lvm2\lvm2-udeb_2.03.02-3_amd64.udeb (624.7 KB)
Extracting: F:\pool\main\l\lvmcfg\lvmcfg_1.50_all.udeb (3.4 KB)
Extracting: F:\pool\main\l\lvmcfg\lvmcfg-utils_1.50_all.udeb (103.4 KB)
Extracting: F:\pool\main\l\lzo2\liblzo2-2-udeb_2.10-0.1_amd64.udeb (40.8 KB)
Extracting: F:\pool\main\liba\libaio\libaio1_0.3.112-3_amd64.deb (11.0 KB)
Extracting: F:\pool\main\liba\libaio\libaio1-udeb_0.3.112-3_amd64.udeb (3.3 KB)
Extracting: F:\pool\main\libb\libbsd\libbsd0-udeb_0.9.1-2+deb10u1_amd64.udeb (36.2 KB)
  Unable to create file: Windows error code 0x000001B1 (FormatMessage error code 0x00003AFC)
Unable to create F:\syslinux.cfg - booting from USB will not work

0 devices found
0 devices found
0 devices found

```

pbatard, 2021-06-09:
I cannot replicate your issue. And error code `0x000001B1` is `ERROR_NO_SUCH_DEVICE` indicating that your device got disconnected, which is further evidenced by the fact that you get `0 devices found` after the error. Rufus does not have the capability to disconnect devices, so this is clearly something external.

Did you actually run a bad blocks ckeck or try a different USB Flash Drive? Also, does the error always happen at the same point?

All in all, it looks like an environmental error from your system or your hardware, and therefore not something that can be addressed by Rufus. I am therefore afraid that you will have to identify the cause of this issue on your own...

<details>
<summary>Click the arrow to see my log</summary>

```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19043.985)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB device 'Kingmax Kingmax USB Device' (1687:0903) [ID]
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3951, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {34F8DA5F-5CF2-49B7-B3EE-B56C3AA4A6B4}
Max parts: 128, Start Offset: 17408, Usable = 32505821696 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'Tails'
  ID: {3141B8BC-B6F5-4641-8FB3-879A5444994A}
  Size: 8 GB (8589934592 bytes)
  Start Sector: 2048, Attributes: 0xD000000000000005
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20190226 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'd-live 10.9.0 gn amd64'
  Size: 2.5 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: debian-live-10.9.0-amd64-gnome.iso (2.5 GB)
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 8 GB)
Opened \\.\PhysicalDrive5 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
[... 994 lines cut ...]
Extracting: F:\.disk\udeb_include (49 bytes)
Created: F:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico

Found USB device 'Kingmax Kingmax USB Device' (1687:0903) [ID]
Using autorun.inf label for drive F: 'd-live 10.9.0 gn amd64'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3951, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02ACCF86
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 30.3 GB (32504807424 bytes)
  Start Sector: 2048, Boot: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.
```
</details>

LeoLTM, 2021-06-09:
Thanks for your reply! 

I had this issue with the first and the second drive, but trying it another time (with the second drive) got it working somehow... The weird thing is that the first USB drive stopped working at exact the same point everytime but has no bad sectors... It's a 64GB Gen3 one.
But I don't wanna bother you with those things, I'll just keep in mind that the 64GB one is faulty ;) 

Thank you for keeping this great project alive!

closed 2021-06-09 by LeoLTM