pbatard/rufus #1750: rufus doesn't work with windows 7 isos on modern or old devices

jredfox, 2021-06-09:
the iso straight from microsoft
https://download.microsoft.com/download/0/6/3/06365375-C346-4D65-87C7-EE41F55F736B/7601.24214.180801-1700.win7sp1_ldr_escrow_CLIENT_PROFESSIONAL_x64FRE_en-us.iso

issue: it boots but, then freezes before the windows 7 logo fully does it's animation.... 

I remember thousands of people using usb recovery as an option for windows 7 so it's got to be a program issue not windows os issue.  **It appears to be an issue with your UEFI :NTFS boot loader of yours**

what I have tried:
GPT partition
MBR partition
enabling and disabling legacy boot
downloading an earlier version of windows 7 iso that can be compatible with fat32 format. that seemed to work but, I only got it working in legacy boot using an older windows 7 but, only on legacy boot which isn't really suppose to boot windows 7
changed the cluster size to 16kb instead of default

yes I:
had secure boot off
tried multiple usb drives (lexar and sandisk)
changed the boot order to prefer usb drives first
waisted all day trying to get rufus working with different configurations and bios configurations

```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19042.985)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7608, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {53779454-CF07-4FF0-889A-02A628B5A028}
Max parts: 128, Start Offset: 17408, Usable = 62578981376 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {3DB71681-DF1D-4E1A-887C-05DFED100286}
  Size: 58.3 GB (62576865792 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {5A90D381-75C8-479C-AB54-321CF401BC73}
  Size: 512 KB (524288 bytes)
  Start Sector: 122222489, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'DVD_ROM'
  Size: 5.5 GB (Projected)
  Has a >4GB file
  Uses: EFI (win7_x64)
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: 7601.24214.180801-1700.win7sp1_ldr_escrow_CLIENT_PROFESSIONAL_x64FRE_en-us.iso (5.5 GB)
WIM extraction method(s) supported: wimgapi.dll
WIM apply method supported: wimgapi.dll

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 58.3 GB)
● Partition 2 (offset: 62577914368, size: 512 KB)
Opened \\.\PhysicalDrive1 for exclusive write access
[... 945 lines cut ...]
   3453498 allocation units available on disk.
Total duration: 179.14 milliseconds (179 ms).
NTFS Fixup completed.

Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7608, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {3C0E1A93-4260-47AE-AA09-97D19FEF7BE5}
Max parts: 128, Start Offset: 17408, Usable = 62578981376 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {5CAE0FD9-1C9E-4861-B14F-CB4C987AA919}
  Size: 58.3 GB (62576865792 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {A1D4DFFA-4336-4B66-B273-56F4AAC7DCD8}
  Size: 512 KB (524288 bytes)
  Start Sector: 122222489, Attributes: 0x0000000000000000
```
LOG: where do you find your logs?

pbatard, 2021-06-09:
> I remember thousands of people using usb recovery as an option for windows 7 so it's got to be a program issue not windows os issue.

I don't quite follow your logic there, especially as the vast majority of people would have been installing Windows on __old__ platforms, that are more likely to actually be compatible with Windows 7 than recent ones. One thing to bear in mind for instance is that Windows 7 has no native support for a lot of the modern hardware, starting with USB 3.0.

> It appears to be an issue with your UEFI :NTFS boot loader of yours

I'm afraid I can disprove that (see below).

But for one thing, you need to realize that all the UEFI:NTFS boot loader does is load a generic NTFS driver (and, if you do read the UEFI specs, you'll understand that as far as **ANY** UEFI operations are concerned, such as booting, they are 100% file system agnostic, meaning that it really doesn't matter one bit if the file system on which the boot files reside is FAT32, NTFS, ext4, or whatever niche file system you want to use. In other words: NO, FAT32 is not more "special" for boot than NTFS in the UEFI world).

>  I have a screenshot showing the entire drive is only NTFS this is not normal bios booting protical.

Yes, you have a screenshot demonstrating that Microsoft's Disk Manager is doing a poor job at showing all the partitions, as there should be a second FAT32 partition displayed after the NTFS one, that contains the UEFI:NTFS files. You will see that partition listed with `DISKPART`, but not Disk Manager, as Disk Manager seems to "remove" partitions that are smaller than a specific size.

As to this not being normal, well, for one thing, a single NTFS partition is how you would boot a Windows 7 installer for BIOS, since the BIOS bootloaders would reside in the MBR and the PBR (Partition Boot Record). So this is super-normal for BIOS.
And for UEFI, because of what I explained above, this is not abnormal, as, once again, as long as you have an NTFS driver (and more and more motherboards have one... which I'll also talk about below), you can boot whatever file system. So you can very well have a single NTFS or ext4 partitions, with a UEFI bootloader, and there's nothing abnormal about it if you have support for said file system in UEFI.

So please do not confuse "uncommon" for "abnormal". The one thing UEFI specs mandate is that, as opposed to an NTFS or ext4 driver, there should at least be a FAT32 UEFI driver included in the UEFI firmware, which in turn means that, where possible, most people would create a FAT32 based drive, since you are then guaranteed that a UEFI firmware should be able to access that file system for boot. But again, all it takes for UEFI to access a file system for boot is to have a driver for it, so, even if you don't have  a driver for a specific file system, what you can do is use a chain loader located on a small FAT32 partition (which you can place at the end of the drive, because there again there's nothing in the UEFI specs that says that the boot partition has to be the first one) and use a boot loader there to load the file system drive you need and chain load the main UEFI bootloader on the non FAT32 main partition. Well, that is just what UEFI:NTFS does, and, again, if you want to tell me that this is "abnormal", then you're going to have to point me to the part of the UEFI specs that indicates so, because I believe I do have acquired some insight into those, and I have yet to find anything to state that this method should not be used, or even, is likely to cause issues.

> Normal windows boot from bios and even linux I believe have a fat32 partition for the boot loader while your tool does not

It actually does. But you need to use a non-crippled disk partition utility (i.e. not Microsoft's Disk manager) to see it. So, already, we are starting with something that you attribute to an issue with Rufus to an issue with Windows. But again, you don't have to take my word for it: Just use any other partition utility, or even `DISKPART` from the commandline, and see for yourself.

Okay, now to the main point:

> rufus doesn't work with windows 7 isos on modern or old devices

How many devices have you tested?

I have tested a media created from your ISO with Rufus 3.14 on a Dell Optiplex 390 (which is a fairly old UEFI based PC), an ASUS Z87 based PC (a not so recent, but not so old PC) and a NUC i7 based PC (which was only released last year), and I saw no freezing issue with the first two. So, the first thing we can dismiss from the get go is your generic statement that "Rufus doesn't work with windows 7 isos on old devices", because I can __demonstrate__ otherwise, and I also know that there are still a lot of folks using Rufus to reinstall Windows 7 on their old devices.

But what about that (modern) i7 based PC?

Well, there I saw your freezing issue... However, the real nice thing about these modern PCs, which also helps demonstrate what I was stating above about FAT32 not being anything special when it comes to UEFI boot, is that this Intel NUC does actually provide a native UEFI NTFS driver, alongside the FAT32 one.

Which means that we can also create a pure NTFS media, __WITHOUT USING RUFUS AT ALL__, from your ISO, and see if that fares any better. All that's needed is create a single NTFS partition (I even made sure I didn't use Rufus at all to do that) and extract the content of the ISO onto it with something like 7-zip. Then (because Windows 7 __is__ [... cut ...] eezing" matter, you will need to take it to Microsoft... which you can't, since they have declared Windows 7 obsolete.

Or, more succinctly, you probably want to stop trying to install an OS that has been declared obsolete, as you won't get much support for issues like this, be it from Microsoft or from me.

As I believe I have answered your question, and especially considering that, just like Microsoft, I am no longer officially supporting the installation of Windows 7, I will close this issue.

closed 2021-06-09 by pbatard

jredfox, 2021-06-09:
It Is 100% rufus’s fault. It’s not a hardware issue my devices are 100% compatible with windows 7. Bold of you to assume that I even had a 3.0 USB port in my crappy laptop or my older one. Also windows 7 supports 3.0 USB since almost always.

The issue is rufus’s boot loader doesn’t really boot because they are doing something wrong in older or newer bios’s then the one you coded it on. Most people only download this because etcher said to use windows xp and windows 7 isos to your program which doesn’t seem to work on a lot of devices.

If you look at ubuntu latest version they have an EFI Fat32 partition all by itself as the boot loader then it boots another format on another partition. I am not saying 100% that this is the issue all I know is windows 7 100% works on the windows 7 pc I tested it on and your usb doesn’t work. Yet the tool from Microsoft for windows 7 does.

If you look at your issues half of them are isos failing to boot properly when they can use another program to work just fine for them but, it’s “hardware issue” when other programs clearly work. The 300 issues you closed within 2 years had nothing to do with an actual issue just people faulsy claiming on not understanding how to run a simple gui program. Listen to yourself that’s not a programmer that’s just stupidity.

I will request etcher to support windows images as your program doesn’t function. If I wanted to only do Linux isos I would use etcher as I doubt you also support mac isos.

pbatard, 2021-06-09:
> It Is 100% rufus’s fault.

Then demonstrate it.

> Bold of you to assume that I even had a 3.0 USB port

I'm not assuming anything. I'm just stating that Windows 7 and USB 3.0 is a common issue for users (which may or may not include you).

>  Also windows 7 supports 3.0 USB since almost always.

[Through proprietary drivers that you need install separately]](https://social.technet.microsoft.com/Forums/windows/en-US/acd63e62-0dbd-4732-9292-7cf7b67a4fe6/usb-30-support-in-windows-7?forum=w7itprohardware). Notice how I used "native" when I talked about USB 3.0 support? That's not by mistake.

>  all I know is windows 7 100% works on the windows 7 pc I tested it on 

Using that specific ISO?

> Yet the tool from Microsoft for windows 7 does.

Using that specific ISO?

> Listen to yourself that’s not a programmer that’s just stupidity.

I get it. You are angry. And, in a few hours, you have somehow have acquired enough insight to declare that 300 issues I closed were actually caused by my application, and not something else.

So you can try to calm down and assess your issue, as I did. Or you can try continuing to insult me, and see how far that takes you.

jredfox, 2021-06-10:
"UEFI supports Fat32 only. UEFI and BIOS booting (Distro dependent)." from https://www.pendrivelinux.com/yumi-multiboot-usb-creator/ . While the iso windows 7 file has files that are bigger then 4GB. so you have to have a seperate boot loader with a parition in there of fat32 from there you can boot off of the ntfs. otherwise it's bios boot only

pbatard, 2021-06-10:
> UEFI supports Fat32 only

__WRONG__

Please point to the [UEFI specs](https://uefi.org/specifications) section that says that UEFI supports FAT32 only.

This is such a [common misconception that I have a dedicated FAQ entry about it](https://github.com/pbatard/rufus/wiki/FAQ#Blah_UEFI_Blah_FAT32_therefore_Rufus_should_Blah) (please read it).

And before you tell me _"What should you know about UEFI specs?", you may want to be made aware that I am a [regular contributor to EDK2](https://github.com/tianocore/edk2/commits?author=pbatard) which is the basis of the UEFI implementation, so, yeah, I believe I know a thing more or two than someone who's propagating a common misconception about FAT32 and UEFI.

> so rufus EUFI:NTFS Boot loader should have FAT32 still on a separate partition then use NTFS for the rest.

Which if you simply had __read__ what I wrote above, you would know it does.

Once again, Disk Manager does not show the FAT partition __because of a Microsoft limitation/bug__. But that partition is there.
Heck, the log from Rufus, which you copy/pasted, tells you it's there:
```
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {5CAE0FD9-1C9E-4861-B14F-CB4C987AA919}
  Size: 58.3 GB (62576865792 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {A1D4DFFA-4336-4B66-B273-56F4AAC7DCD8}
  Size: 512 KB (524288 bytes)
  Start Sector: 122222489, Attributes: 0x0000000000000000
```

> As shown in the image above everything was NTFS partitioning not fat32

As shown by using a different partition utility than Windows Disk Manager, you can not trust what you see from that image, as it is not representing the true layout of the disk. That's because, for reasons that only Microsoft knows, it is not showing the extra FAT partition that exists at the end of the disk, that any other partition utility will show is there.

I __told you that__ from the get go, but you are completely ignoring what I am telling you, because you appear to have long decided that Rufus has to be doing something wrong, and/or that other components, such as Microsoft software, can't. Sadly for you, that is far from the truth.

So if you are not going to start reading what I am explicitly telling you, I'm just going to stop right there.

> Either way you or etcher add support

Rufus already has support for your image. It works perfectly fine on the hardware I tested that's old enough to support Windows 7 and, as you should be able to see (since you're supposed to have spent time perusing 200 closed issues), I'm not getting that many complaints from people unable to use Rufus to (re)install Windows 7, which has to mean that I don't really need to do anything.

Sorry pal, but your issue is environmental, and it pertains to the installation of an obsolete OS, so it's 100% on you to fix it.

Ans since I don't foresee anything good coming up from someone who, clearly, is unable to read what's being said to them, and who is ordering developers of software that they got absolutely for free to drop what they are doing to solve their specific environmental issue, I am both locking this issue and banning you.