pbatard/rufus #1751: Future request: Build in support for kickstart parameter for Linux

ohrly, 2021-06-18:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Enhancement description
-----------------
Hi, great tool, thanks.
I like to see the support for the "inst.ks=" or "ks=" parameter for grub.cfg and isolinux.cfg. #You already change the "inst.stage2=" parameter to "hd:LABEL=...."!
If you also change the kickstart parameter there is no need to change anything from an kickstart ISO. Could make life easier... ;-)

No Logs for this. This part (isolinux.cfg):
```
label check
  menu label Test this ^media & install RHEL 8.1.0
  menu default
  kernel vmlinuz
  append initrd=initrd.img inst.stage2=hd:LABEL=RHEL-8-1 rd.live.check inst.ks=hd:LABEL=RHEL-8-1:/base.ks quiet

label linux
  menu label ^Install RHEL 8.1.0
  kernel vmlinuz
  append initrd=initrd.img inst.stage2=hd:LABEL=RHEL-8-1 **inst.ks=hd:LABEL=RHEL-8-1:/base.ks** quiet
```

pbatard, 2021-06-18:
> You already change the "inst.stage2=" parameter to "hd:LABEL=...."!

No we don't.

We do change the label, because of the FAT32 limitations that restrict the label to uppercase and 11 characters, which means that, if we have a config file that has `LABEL=SomeDist 1234`, then we're going to alter this to `LABEL=SOMEDIST%2012` so that the installed can locate the media. But the only part being altered here is the label. Especially, we do not change an `inst.stage2=` parameter to `hd:LABEL=`.

The goal in Rufus is to only change what we don't have any choice but to change, and leave everything else as is, since our mandate is to have the USB behave the as close a possible as the original ISO.

But the nice thing is that, since Rufus usually allows you to edit the files on the media (as opposed to utilities like Etcher), if you do want a specific config file option, you can always alter it yourself.

Then again, if you are using a recent Red Hat image, this request makes little sense anyway, since Red Hat __broke__ ISO mode, and thus the only way to write a recent Red Hat image right now is through DD. But ultimately this means that Rufus doesn't have the capability to edit anything for you anyway.

&rarr Rejected.

closed 2021-06-18 by pbatard

ohrly, 2021-06-18:
ok, sorry. You are right. I did change that and did not remember it. Than the workaround will be enough.