pbatard/rufus #1753: add new window

LievreAI, 2021-06-20:
hey guys i want to add a new windows to the application like windows server 2016

pbatard, 2021-06-20:
The Windows downloads proposed by Rufus are handled by the [Fido script](https://github.com/pbatard/Fido).

I happily accept patches for the Fido script if you have any.

Note that the ISOs being proposed must come from a public link offered by Microsoft, such as the ones from https://www.microsoft.com/en-gb/software-download/windows10ISO. Are the Windows Server 2016 ISOs publicly available?

pbatard, 2021-06-22:
Since this is a Fido issue, and OP has not replied to the public availability of Windows Server 2016 ISOs, I will close this issue.

closed 2021-06-22 by pbatard