pbatard/rufus #1754: 3.14 report error "could not find procedure xxx" in Win7_X64

huntagain2008, 2021-06-24:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
rufus 3.14 is not available in win7_X64, burning ISO files initially reported an error, "can not find procedure file"
Switching to rufus 3.9 to burn the ISO file was successful.

Sorry about my English. I am from China.It is my first time to use rufus.

Log
---
```
<FULL LOG>
```

pbatard, 2021-06-24:
1. I have had sporadic reports of this issue, but I have tried on multiple Windows 7 machines, and I am unable to replicate it at all
2. **PLEASE** provide the log as requested. I cannot help you unless you provide a log.
3. I will need you to tell me what is the last version of Rufus, before version 3.14, that appears to work. There have been 4 versions between 3.9 and 3.14, that you can download at https://rufus.ie/downloads/, and at the very least, I need to know if the problem was introduced between 3.9 and 3.10, or 3.10 and 3.11, or 3.11 and 3.12, etc.

Considering that I am unable to replicate the issue, unless you can provide the log and also tell me between which version the problem seems to have been introduced, I will have no choice but to close this issue, as I won't have enough data to troubleshoot it.

huntagain2008, 2021-06-25:
@pbatard 
Because my English is not good, the following text is translated with translation software.

I'm running 3.14rufus on an old Dell laptop from 2014, and I got a pop-up error "can not find procedure" before the iso burning step.

I'm sorry, I forgot to save the log at that time. At that time I couldn't find a solution for this error by searching online. Finally I had to use the old rufus 3.9 to create a USB boot disk successfully. I saw at that time that most Chinese sites used rufus 3.9, so I thought 3.9 was more stable, so I switched to rufus 3.9 to run it, and it worked fine.

The OS of that laptop at that time was installed with a Chinese PE installer with malware and the ISO file was Win7_ghost. I couldn't find a good U disk boot creation tool at that time. Thinking of reinstalling the laptop's system as the original again. I ended up reinstalling that Win7 to Win10LTSC2019 using rufus 9.0. This time the system was very clean and fast.

I only ask this question to show that I am using the rufus you developed and want to show that the software works well. In fact, just ran rufus 3.14 completely fine on my own desktop with Win10.

pbatard, 2021-06-25:
In that case I have no choice but to close the issue and wait for someone to provide the information I requested.

closed 2021-06-25 by pbatard