pbatard/rufus #1756: Rufus Error [0x00000070]

Officialtbh, 2021-07-04:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ]x I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
  - [ x] Plugging the USB into a different port.

Issue description
-----------------
Im having an error [0x00000070] but i have made sure my flash drive is completely empty and working correctly. Its completely empty but rufus is still telling me that i do not have enough disk space. Also, ive used the same empty flash from rufus once before and it didnt give me any problems like it is now. Does anyone know why / how to fix it? Im not skilled with computers so ive tried everything i can do with my knowledge.

<FULL LOG>  Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19042.1052)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
Downloading https://github.com/pbatard/Fido/releases/download/v1.20/Fido.ps1.lzma
File length: 13.6 KB
Successfully downloaded 'Fido.ps1.lzma'
Signature is valid ✓
Please Wait...
Exited download script with code: 100
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
The VHD size is too large for the target drive
0 devices found
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
Could not open GUID volume '\\?\Volume{ad007837-da2c-11eb-9558-c858c0f23ee6}': [0x00000005] Access is denied.
No logical drive found (unpartitioned?)
1 device found
Could not open GUID volume '\\?\Volume{ad007837-da2c-11eb-9558-c858c0f23ee6}': [0x00000005] Access is denied.
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
Could not open GUID volume '\\?\Volume{ad007837-da2c-11eb-9558-c858c0f23ee6}': [0x00000005] Access is denied.
No logical drive found (unpartitioned?)
1 device found
Could not open GUID volume '\\?\Volume{ad007837-da2c-11eb-9558-c858c0f23ee6}': [0x00000005] Access is denied.
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
0 devices found
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No

pbatard, 2021-07-04:
> Im having an error [0x00000070] (...) rufus is still telling me that i do not have enough disk space

Please provide a log that shows these errors.
The log you attached does not show any of that. It doesn't even show what image you selected and what happens when after you press `START`.

If you want to get help, please make sure that your log shows what happens when you are trying to accomplish the operation  that results in the errors you mention.

Unless you provide such a log, I genuinely cannot help you. So please retry what you were doing when you got these errors, and make sure you attach the full log you get after that.

Officialtbh, 2021-07-04:
i will try to get the correct log. From what i remember i couldnt pull up the log while the error messege was being displayed, so i closed it and went to the log straight afterwards. I have started to attempt to download load it again and will update in around 20 minutes.

Officialtbh, 2021-07-04:
for some reason it successfully downloaded. but now im getting  the error "unable to assign drive letter." @pbatard 

<FULL LOG> Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19042.1052)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
Downloading https://github.com/pbatard/Fido/releases/download/v1.20/Fido.ps1.lzma
File length: 13.6 KB
Successfully downloaded 'Fido.ps1.lzma'
Signature is valid ✓
Please Wait...
Exited download script with code: 100
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
The VHD size is too large for the target drive
0 devices found
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
Could not open GUID volume '\\?\Volume{ad007837-da2c-11eb-9558-c858c0f23ee6}': [0x00000005] Access is denied.
No logical drive found (unpartitioned?)
1 device found
Could not open GUID volume '\\?\Volume{ad007837-da2c-11eb-9558-c858c0f23ee6}': [0x00000005] Access is denied.
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
[... 280 lines cut ...]
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No

Format operation started
Requesting disk access...
ABORTED: Cannot use an image that is located on the target drive!
Unable to find a drive letter to use

Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x480EE298
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.9 GB (31010979840 bytes)
  Start Sector: 36096, Boot: No

pbatard, 2021-07-04:
1. You don't need to download a Windows ISO every time. You are supposed to download it __once__ and then, if you want to reuse it, you just use the `SELECT` button to locate and select that download.
2. `ABORTED: Cannot use an image that is located on the target drive!` is the actual issue you need to worry about. It's telling you that you downloaded the ISO file onto the USB drive. But the USB will be completely erased, and the ISO file is still needed to create the bootable media, so, if Rufus was to continue, it would delete that ISO, and then be unable to use it for the media creation. So it is telling you that this will not work, as it's like sawing the branch you are sitting on. In short, you need to make sure that the ISO is __NOT__ on the USB drive you are trying to make bootable using Rufus. For instance, you need to make sure that the Windows ISO is located on your `C:` drive. If you do that, I'm pretty sure you'll find that things work a lot better...

Officialtbh, 2021-07-04:
How would one do that? I moved the file onto my C drive but its still trying to push it through my D drive, and i have a feeling that if i delete it off the D drive it wont work either? Im not good with software so i apologize if i am frustrating you.

pbatard, 2021-07-05:
I would advise you to watch youtube videos on how to use Rufus to install Windows, because you probably need to be shown rather than told what to do.

Again, you __must__ move that ISO outside of the `D:` drive for things to work, because it's not up to you to copy the content of the ISO to the USB (since doing it needs to be transformed for that to work), but it's Rufus' job.

You don't delete it either. You just make sure that you move it from the `D:` drive to somewhere on your `C:` drive. And then you click `SELECT` in Rufus to look up that ISO and select it, from the `C:` drive.

Please don't try to "guess" how things should work. Just try to follow what we are telling you to do, even if it goes against what you think should work. And, again, search on youtube for videos on how to install Windows with Rufus, as it should make things a bit clearer.

Officialtbh, 2021-07-06:
Alright, so i did some research and got past a good few steps. Though now i cant get past the "disable secure boot" process. So i disabled it but rufus is still giving me a message telling me to disable it. ive been looking for a solution for about an hour or two but couldnt find anything. I got around the drive letter part by copying the ISO from the D drive to the C, though im not sure if i should not delete it off the D drive? Maybe thats whats happening. Ill keep looking for a solution for this issue and if i find one ill either post again or edit this post. Ill also include the log again, not sure if its needed or not.

<FULL LOG>
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19042.1052)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {3B44CA6F-542A-4594-BF49-E0CF5049598C}
Max parts: 128, Start Offset: 17408, Usable = 31029426688 bytes
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {3B44CA6F-542A-4594-BF49-E0CF5049598C}
Max parts: 128, Start Offset: 17408, Usable = 31029426688 bytes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 5 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1909_English_x64 (1).iso (5 GB)

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
No partition to delete on disk
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 28.9 GB)
● Creating UEFI:NTFS Partition (offset: 31028353024, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{bf18c6b0-de3f-11eb-955a-c858c0f23ee6}\
Disabling file indexing...
Successfully remounted \\?\Volume{bf18c6b0-de3f-11eb-955a-c858c0f23ee6}\ as D:
Extracting files...
Image is an UDF image
Extracting: D:\autorun.inf (128 bytes)
[... 2695 lines cut ...]
Extracting: D:\sources\replacementmanifests\legacy-sapi-repl.man (4.7 KB)
Extracting: D:\sources\replacementmanifests\mdm_migration-replacement.man (25.0 KB)
Extracting: D:\sources\replacementmanifests\media-settings-migration-replacement.man (993 bytes)
Extracting: D:\sources\replacementmanifests\mfmpeg2srcsnk-migration-replacement.man (1 KB)
Extracting: D:\sources\replacementmanifests\mfplat-migration-replacement.man (1.1 KB)
Extracting: D:\sources\replacementmanifests\mfsrcsnk-migration-replacement.man (1.1 KB)
Extracting: D:\sources\replacementmanifests\microsoft-activedirectory-powershell-replacement.man (703 bytes)
Extracting: D:\sources\replacementmanifests\microsoft-activedirectory-webservices\adwsmigrate.dll (100 KB)
Extracting: D:\sources\replacementmanifests\microsoft-certificateauthority-enrollment-serverupgrade-replacement.man (1.9 KB)
Extracting: D:\sources\replacementmanifests\microsoft-certificateservices-ca-serverupgrade-replacement.man (2 KB)
Extracting: D:\sources\replacementmanifests\microsoft-certificateservices-mscep-serverupgrade-replacement.man (2.0 KB)
Extracting: D:\sources\replacementmanifests\microsoft-certificateservices-ocsp-serverupgrade-replacement.man (1.9 KB)
Extracting: D:\sources\replacementmanifests\microsoft-certificateservices-policy-serverupgrade-replacement.man (1.9 KB)
Extracting: D:\sources\replacementmanifests\microsoft-certificateservices-webenrollment-serverupgrade-replacement.man (2 KB)
Extracting: D:\sources\replacementmanifests\microsoft-client-license-platform-service-migration\clipmigplugin.dll (242.3 KB)
Extracting: D:\sources\replacementmanifests\microsoft-hyper-v-client-migration-replacement.man (1.3 KB)
Extracting: D:\sources\replacementmanifests\microsoft-hyper-v-drivers-migration-replacement.man (634 bytes)
Extracting: D:\sources\replacementmanifests\microsoft-hyper-v-migration-replacement.man (5.8 KB)
Extracting: D:\sources\replacementmanifests\microsoft-onecore-tiledatarepository\tilestoremigrationplugin.dll (111 KB)
Extracting: D:\sources\replacementmanifests\microsoft-onecore-tiledatarepository-replacement.man (918 bytes)
Extracting: D:\sources\replacementmanifests\microsoft-windows-advertisingid-replacement.man (1.2 KB)
Extracting: D:\sources\replacementmanifests\microsoft-windows-appmanagement-migration\appmanmigrationplugin.dll (1.1 MB)
Extracting: D:\sources\replacementmanifests\microsoft-windows-appx-deployment-server\appxprovisioning.xml (21.2 KB)
Extracting: D:\sources\replacementmanifests\microsoft-windows-appx-deployment-server\appxupgrademigrationplugin.dll (440.5 KB)
Extracting: D:\sources\replacementmanifests\microsoft-windows-appx-deployment-server\microsoft.windows.secondarytileexpe

pbatard, 2021-07-06:
> but rufus is still giving me a message telling me to disable it.

Rufus will give that message whether Secure Boot is disabled or not. Please don't expect Rufus to tell you what the Secure Boot state of your current machine is, because it doesn't do that. In other words, it's not because you see Rufus telling you that you have to disable Secure Boot that Secure Boot is still enabled.

It's more like the "remember to fasten your seatbelt" sign that you may see on the road. You may see those signs, even if you have your seatbelt on, as it's just a __reminder__.

> I got around the drive letter part by copying the ISO from the D drive to the C

Good.

> though im not sure if i should not delete it off the D drive? 

Don't worry about it. Rufus will delete it anyway, so you don't have to do it yourself.

As far as I am concerned (and from what I can see from your log) you USB is all set to install Windows.
So, if you disabled Secure Boot, you just need to plug it in into the machine you want to install Windows on, make sure that your BIOS/UEFI options are set to boot from USB first (but you really only need to check that if you find that your machine doesn't boot into the Windows installer) and see what happens.

Officialtbh, 2021-07-06:
So everythings good to go, that means i just have to get into BIOS and set my start up to boot from my USB? Ill attempt it now to see what happens. It should just start up like a brand new computer would right? Were it shows that welcome / setup this computer type of screen?

pbatard, 2021-07-06:
Something like that. But please don't expect me to guide you through the Windows installation process or even through the boot options of your BIOS/UEFI in case you have trouble booting from USB, as this is well outside of the scope of what I can provide support for.

In fact, since the log shows you did manage to create a bootable Windows installation USB, and the error your reported were not actaul Rufus problems, I will close this support ticket. If you need more help with this, please ask in Windows support forums or in your computer manufacturer's support forums.

closed 2021-07-06 by pbatard

Officialtbh, 2021-07-06:
Thank you for the help! Managed to get the bios going and now Windows is doing its thing. Learned some software stuff along the way (even though it was very basic) and hopefully can learn more along the way here.