pbatard/rufus #1757: Rufus it's wayy too slow, and sometimes doesn't open (Problem with SD Reader)

elalann, 2021-07-11:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Since I installed this generic internal 2.5" on my computer, Rufus started to take really long to open, and sometimes just doesn't open. If I use DebugView, I can see exactly the same of the log, it's detecting and removing the device over and over and over.

When it open, it works... okay. Not as fast as it used to be, not really sure if it keeps checking or what.
Weird enough, it also takes longer on "checking for conflicting devices" (not gonna mix issues or make one for that, because isn't a deal breaker, only 15-30s more)

I can confirm Rufus working as lovely as it used to work on my computer before, on another notebook I have laying around.
NOTE: I'm not trying to work with this sd card reader, instead I'm making an usb3 bootable.

Log
---
```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 17763.1999)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic- MS/MS-Pro/HG USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SM/xD-Picture USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC/MS/MSPRO USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- Compact Flash USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
0 devices found
Found USB 2.0 device 'Generic- MS/MS-Pro/HG USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SM/xD-Picture USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC/MS/MSPRO USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- Compact Flash USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1884, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {4B73007D-7B72-4D24-BBE5-5EEA82ECF26E}
Max parts: 128, Start Offset: 17408, Usable = 15502113280 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {4B5516FB-60D2-42EB-81E8-D1A85CBC6862}
  Size: 14.4 GB (15501082112 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Checking for Rufus updates...
Checking release channel...
No new release version found.

```

pbatard, 2021-07-11:
I'm afraid that this looks like a pure hardware issue to me.

For one thing, you are saying that _"it's detecting and removing the device over and over and over"_, which can only mean that Windows is sending a hotplug connection event over and over (which is how Rufus gets notified that it should re-enumerate devices). And, before you ask, yes, we do limit the rate of how frequently Rufus should handle hotplug events, to less than 1 per second (which is what we want to detect manual insertions and removals done by users). Therefore, if you are seeing your device being detected over and over, even as we limit the rate, it means that __Windows__ (not Rufus) is sending an unnatural number of hotplug events, which means that your hardware has an issue.

I'm also a bit puzzled by this excerpt from the log:  
```
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
```
But I'm going to assume that, on the machine where you installed this new SD card reader, you only have USB 2.0 ports (which would make it a rather old computer). If that is not the case, then this would further corroborate the idea that your USB bus is not behaving as expected, due to a pure hardware issue.

Well, I'm afraid Rufus cannot be expected to perform miracles or even work around problematic hardware. For one thing, the most likely outcome, if I were to add a custom workaround to speed up detection for flaky devices, is that drive creation would still fail, or induce extended write delays, and then I'd get a new report of _"Now Rufus is very slow to create the media. Fix this!!!"_, as it'd still boil down to someone's hardware not performing as it should.

In other words, sorry, but I am not going to do anything about something that appears to stem from using flaky hardware, as Windows hotplug events should not happen in the manner you see them happen if the hardware was working properly. The cost/benefit ratio of trying to apply a workaround for a specific piece of malfunctioning hardware is simply too high, and I genuinely don't have the time to try to work around punctual hardware issue that is only affecting a handful of users at best (and yes, I can be affirmative that this isn't affecting a large number of users, on account that Rufus is downloaded more than 1 million times every month and that yours wouldn't be the first report I am seeing of device detection and removal if that was the case). The only other instances I know of slow detection, that aren't caused by flaky hardware, have to do with using an Antivirus solution like Sophos, but they do not produce the kind of log that you got.

In other words, from where I stand, this new SD reader device you installed is not working properly and you will need to fix that if you want Rufus to work with it.

As such, I will close this issue.

closed 2021-07-11 by pbatard

elalann, 2021-07-12:
Hello Pete,
No worries, I wasn't even pretending you or someone to put effort on something that (probably) affects me with this specific hardware combination.

As for the line of the USB drive, you're mixing stuff. It's only because im lazy af and plugged in the usb on an 2.0 port, nothing else.

I opened the issue only for informational purposes, maybe it helped somehow to fix another thing or it may help in any way this gets referenced later in the time.

To be honest, I'm not on software development, and I don't know how Windows manages and works with drives and so on, It's not my area. I thought it was Rufus who checked for devices, saw one removed and tried again.
The SD card reader, it's a cheap POS, I'm gonna post this comment and then faster than the speed of light throw that thing into the bin.

Hope you have an awesome day!
Regards,
Alan.